analogdevicesinc/libiio #8: iioxml.py crashes wtih null pointer exception

beOn, 2015-03-28:
Thanks for this great project! I'm especially excited for the python bindings, and have been trying to get them to work today.

Is iioxml.py expected to work at this point? If not, please disregard. If so... I still could have messed something up, but I think I have everything installed correctly. When I try to run iioxml.py, I get this:

```
Traceback (most recent call last):
  File "./examples/iioxml.py", line 83, in <module>
    main()
  File "./examples/iioxml.py", line 66, in main
    context = LocalContext()
  File "/home/pi/devdir/tmp/libiio/bindings/python/iio.py", line 673, in __init__
    ctx = _new_local()
  File "/home/pi/devdir/tmp/libiio/bindings/python/iio.py", line 25, in _checkNull
    raise Exception("Null pointer")
Exception: Null pointer
Exception AttributeError: "'LocalContext' object has no attribute '_context'" in <bound method LocalContext.__del__ of <iio.LocalContext object at 0xb6d002b0>> ignored
```

I get the same whenever I try to initialize a LocalContext object. Is it working for you? If so, and you can't reproduce the issue, let me know and I'll do some more thorough reporting.

rgetz, 2015-03-29:
Ben:

All the examples are expected to work. 

We have also had others report it working. 

https://ez.analog.com/mobile/mobile-access.jspa#jive-content?content=%2Fapi%2Fcore%2Fv3%2Fcontents%2F127137

So, steps to reproduce would be great. 

Thanks
-Robin

beOn, 2015-03-29:
Alright - will do.

beOn, 2015-03-30:
So here's my setup. The device is a Raspberry Pi B+, and I'm running Raspbian GNU/Linux 7 (wheezy). I have enabled i2c on the device, and if I print the list of devices, I can see the AD5254 I'm trying to talk to:

```bash
$ i2cdetect -y 0
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- -- 
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- 2d -- -- 
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
70: -- -- -- -- -- -- -- --  
```

I installed all of the libiio dependencies using apt-get, then installed libiio per the instructions in "what_is_libiio.pdb". No issues (or at least no complaints). But then if I try to run iio_info, I get:

```bash
$ iio_info
Library version: 0.5 (git tag: 6399de7)
ERROR: Unable to create IIO context
```

And if I run iioxml.py, I get the error mentioned above. If there's some specific C you'd like me to run, or anything else I can do to generate some more helpful information (I realize that what I've included here is pretty paltry), please let me know what specifically would be helpful, and I'll be happy to oblige!

rgetz, 2015-03-30:
Ben:

Is the AD5254 driver installed on your PI?

Plus - I'm not sure the dpot driver has been updated for IIO yet. 

http://wiki.analog.com/resources/tools-software/linux-drivers/misc/dpot

- Robin

beOn, 2015-03-30:
No, no sir it is not. I'd be happy to test it out... but we're getting a little out of my wheelhouse here. Sorry! Just to expose a bit of my ignorance: do I need to rebuild the kernel? Is there a handy dandy guide somewhere to installing the driver, or similar drivers? If you point me in the right direction, I'll be happy to do the work!

pcercuei, 2015-03-30:
About the Python bindings - I seriously enhanced them a few days ago, so that they provide all the functionalities of the C library. I still have to update the iioxml.py example.

beOn, 2015-03-31:
Still no luck! I have the driver successfully installed (I can see all of the eeproms and rdacs in /sys/bus/i2c/devices/0-002d), but still get the same exception. Well... one weird thing - I can read the eeproms and rdacs, but writing to them doesn't work (permissions error, even if I use sudo). I'm asking for a little help on that over in the linux drivers forum (https://ez.analog.com/message/181971#181971).

Would you expect this to cause the exception I'm getting? If not, do you have any quick suggestions on steps to generate a more informative error message?

@pcercuei - Thank you very much for the fine work. Can't wait to use them!

pcercuei, 2015-04-01:
Problem solved: I deleted iioxml.py :)
The reason is explained in the commit message.
But I added a new example program, iio_info.py.

beOn, 2015-04-08:
Alright, got the driver working, I can read to and write from all of the sysfs attributes. Updated my local repo, cleaned and rebuilt... still:

```
$ iio_info 
Library version: 0.5 (git tag: b6ad56d)
ERROR: Unable to create IIO context
```

pcercuei, 2015-04-09:
So do you have ```/sys/bus/iio/devices/iio:device0```?

pcercuei, 2015-05-19:
I will close this issue now since it is highly unlikely to be a bug from the libiio library, and it's been one month since the last update.

closed 2015-05-19 by pcercuei

jadonk, 2021-01-11:
I have a new iio device driver that seems to be creating a similar issue: https://github.com/Seeed-Studio/seeed-linux-dtoverlays/tree/master/modules/hm3301

The SYSFS entries work fine, but libiio returns errors trying to create the local: context.

beOn, 2021-01-12:
Oh man, it has been ages. I forget what the resolution was. -Ben