analogdevicesinc/libiio #10: iio_device_create_buffer always fails

alphasilver, 2015-04-01:
I'm trying to read from ADIS16209 using libiio, but when I create a buffer with **iio_device_create_buffer** I'm always getting a nullptr. I've reviewed my code and the documentation, but can't find any explanation why this is happening.

I've been reading libiio documentation and http://wiki.analog.com/resources/tools-software/linux-software/libiio_internals, but didn't find any guideline on the recommended buffer size. Perhaps anyone can explain me what criteria should be used to set the buffer size.

Thank you for this project.

* **Source code**:

```c++
int main()
{
    //...
    struct iio_context* ctx;
    ctx = iio_create_local_context();

    if (!ctx) return 1;

    read_device(ctx, "adis16209");

    iio_context_destroy(ctx);
    return exit_code;
}
void read_device(iio_context* ctx, string dev_name)
{
    iio_device* dev = get_device(ctx, dev_name);
    if(nullptr == dev)
    {
        exit_code = EXIT_FAILURE;
        return;
    }
    cout << "device : " << dev_name << endl;

    // Enable all channels
    unsigned int num_channels = iio_device_get_channels_count(dev);
    for (unsigned int i = 0; i < num_channels; i++)
    {
        iio_channel* chn = iio_device_get_channel(dev, i);
        iio_channel_enable(chn);

        string chn_name;
        {
            const char *name = iio_channel_get_name(chn);
            if (name!=NULL)
                chn_name = name;
            else
                chn_name = iio_channel_get_id(chn);
        }
        bool is_output = iio_channel_is_output(chn);
        cout << " enabled ch" << i << " : " << chn_name << (is_output ? "(output)":"(input)") << endl;
    }

    size_t samples_count=1024*num_channels;

    iio_buffer *buffer;
    buffer = iio_device_create_buffer(dev, samples_count, false);

   if (!buffer)
    {
        cout << "Unable to allocate buffer"<<endl;
        exit_code = EXIT_FAILURE;
        return;
    }

    while (app_running)
    {
        int ret = iio_buffer_refill(buffer);
        if (ret < 0) {
            cout << "Unable to refill buffer" << endl;
            exit_code = EXIT_FAILURE;
            break;
        }

        iio_buffer_foreach_sample(buffer, print_sample, NULL);
        fflush(stdout);
    }

    iio_buffer_destroy(buffer);
    return;
}
```

* **application output**:

device : adis16209
 enabled ch0 : supply(input)
 enabled ch1 : accel_x(input)
 enabled ch2 : accel_y(input)
 enabled ch3 : voltage1(input)
 enabled ch4 : temp0(input)
 enabled ch5 : incli_x(input)
 enabled ch6 : incli_y(input)
 enabled ch7 : rot_x(input)
 enabled ch8 : timestamp(input)
Unable to allocate buffer

* **iio_info output**:

$ iio_info
Library version: 0.5 (git tag: 6399de7)
IIO context created with local backend.
Backend version: 0.5 (git tag: 6399de7)
Backend description string: Linux beaglebone 3.14.35-ti-r55 #1 SMP PREEMPT Fri Mar 27 01:37:50 WET 2015 armv7l
IIO context has 2 devices:
	iio_sysfs_trigger: 
		0 channels found:
		2 device-specific attributes found:
ERROR: Unable to read attribute: add_trigger
ERROR: Unable to read attribute: remove_trigger
	iio:device0: adis16209
		9 channels found:
			voltage0: supply (input)
			5 channel-specific attributes found:
				attr 0: scale value: 0.305180
				attr 1: raw value: 0
				attr 2: en value: 0
				attr 3: index value: 0
				attr 4: type value: be:u14/16>>0
			accel_x:  (input)
			5 channel-specific attributes found:
				attr 0: calibbias value: 0
				attr 1: raw value: 0
				attr 2: index value: 1
				attr 3: type value: be:s14/16>>0
				attr 4: en value: 0
			accel_y:  (input)
			5 channel-specific attributes found:
				attr 0: raw value: 0
				attr 1: calibbias value: 0
				attr 2: type value: be:s14/16>>0
				attr 3: index value: 2
				attr 4: en value: 0
			voltage1:  (input)
			5 channel-specific attributes found:
				attr 0: raw value: 0
				attr 1: scale value: 0.610500
				attr 2: en value: 0
				attr 3: index value: 3
				attr 4: type value: be:u12/16>>0
			temp0:  (input)
			6 channel-specific attributes found:
				attr 0: offset value: -1331
				attr 1: scale value: -470.000000
				attr 2: raw value: 0
				attr 3: index value: 4
				attr 4: en value: 0
				attr 5: type value: be:u12/16>>0
			incli_x:  (input)
			4 channel-specific attributes found:
				attr 0: raw value: 0
				attr 1: index value: 5
				attr 2: type value: be:s14/16>>0
				attr 3: en value: 0
			incli_y:  (input)
			4 channel-specific attributes found:
				attr 0: raw value: 0
				attr 1: type value: be:s14/16>>0
				attr 2: en value: 0
				attr 3: index value: 6
			rot_x:  (input)
			4 channel-specific attributes found:
				attr 0: raw value: 0
				attr 1: type value: be:s14/16>>0
				attr 2: index value: 7
				attr 3: en value: 0
			timestamp:  (input)
			3 channel-specific attributes found:
				attr 0: type value: le:s64/64>>0
				attr 1: index value: 8
				attr 2: en value: 0
		3 device-specific attributes found:
				attr 0: in_accel_scale value: 0.002394195
				attr 1: in_incli_scale value: 0.025000
				attr 2: in_rot_scale value: 0.025000

lclausen-adi, 2015-04-01:
Hi,

It looks as if your ADIS16209 device was registered without a interrupt. For buffered operation the interrupt needs to be connected and setup to be able to notify the driver that a new sample is available.

- Lars

alphasilver, 2015-04-01:
Hi Lars,

Thank you for your feedback.
Can you please give me some hint on how to connect and setup the interrupt? Do you have an example or some document that I can read? 

Thanks

lclausen-adi, 2015-04-01:
Hi,

Depends on you hardware platform. How do you register the ADIS16209 device? If you use a spi_board_info table you set the interrupt number via the irq field. If you use devicetree you can specify the interrupt number using the interrupt property. The interrupt number itself will depend on your hardware platform and needs to match the GPIO pin that the IRQ signal of the ADIS16209 is connected to. For a bit more information check out the ADIS16209 driver documentation at http://wiki.analog.com/resources/tools-software/linux-drivers/iio-accelerometer/adis16209#declaring_spi_slave_devices

alphasilver, 2015-04-01:
Hi,

I'm registering the ADIS16209 in the devicetree:

```c
&am33xx_pinmux {
	/* ... */
	spi0_pins: pinmux_spi0_pins {
		pinctrl-single,pins = <
			0x150 (PIN_INPUT_PULLUP | MUX_MODE0)	/* spi0_sclk.spi0_sclk */
			0x154 (PIN_INPUT_PULLUP | MUX_MODE0)	/* spi0_d0.spi0_d0 */
			0x158 (PIN_OUTPUT_PULLUP | MUX_MODE0)	/* spi0_d1.spi0_d1 */
			0x15c (PIN_OUTPUT_PULLUP | MUX_MODE0)	/* spi0_cs0.spi0_cs0 */
		>;
	};
```

```c
&spi0 {
	pinctrl-names = "default";
	pinctrl-0 = <&spi0_pins>;
	status = "okay";

	spidev0: spi@0 {
		compatible = "adi,adis16209";
		reg = <0>;
		spi-max-frequency = <16000000>;
		spi-cpha;
	};

	spidev1: spi@1 {
		compatible = "spidev";
		reg = <1>;
		spi-max-frequency = <16000000>;
	};
};
```

I'm sorry, but this is my first time doing this, and need a little further help.
Where should I declare the interrupt property (inside spidev0: spi@0 node?), and how can I find which (spi0) pin should match the IRQ signal of the device?

Thank you.

lclausen-adi, 2015-04-01:
The ADIS16209 has a separate pin for the interrupt signal, it's the DIO1 pin. You need to connect this pin to one of the free GPIO pins on your baseboard.

The interrupt property can be added anywhere within the spdev0 node. E.g.:
```c
spidev0: spi@0 {
        compatible = "adi,adis16209";
        reg = <0>;
        spi-max-frequency = <16000000>;
        spi-cpha;
        interrupt-parent = <&gpio1>;
        interrupts = <23 IRQ_TYPE_EDGE_RISING>;
    };
```

interrupt-parent species which GPIO core or bank is the interrupt signal is connected to and the first number in the interrupts property specifies the GPIO pin number within that bank. Both of them need to be modified according to the GPIO pin you chose.

alphasilver, 2015-04-07:
Hi,
I have connected the DIO1 of the ADIS16209 to a GPIO on my board, but the result of iio_device_create_buffer is always the same: nullptr.

Do I need to setup the IRQ elsewhere? I guess that it might be necessary to write to some registers on the device to enable buffered operation, but don't know exactly which ones or how to do it (use iio_device_reg_write?). I see that the default for MSC_CTRL is data-ready disabled - maybe this is the source of my problem.

Thank you for all your help.

**Details:**
I'm using the evaluation board ADIS1620x/PCB with ADIS16209.
I've connected pin 12 on J2 (DIO1) to GPIO0_27 on the Beaglebone black.

According to System Reference Manual of the BBB:
   Head_pin: P8_17
   ADDR/OFFSET: 0x82/02c
   GPIO NO.: 27
   Name: GPIO0_27
   Mode7: gpio0[27]

I rebuilt the device-tree with:
```c
        spidev0: spi@0 {
                compatible = "adi,adis16209";
                reg = <0>;
                spi-max-frequency = <16000000>;
                spi-cpha;

                interrupt-parent = <&gpio0>;
                interrupts = <27 IRQ_TYPE_EDGE_RISING>;
        };
```

pcercuei, 2015-04-10:
What's the value of the "errno" variable after ```iio_device_create_buffer``` returns NULL?
Do you see interrupts happening for your SPI device in ```/proc/interrupts```?

alphasilver, 2015-04-10:
Hi,
After iio_device_create_buffer returning NULL, the errno is **13**.

~ra@beaglebone:/proc$ cat interrupts 
           CPU0       
 23:          1      INTC   7  tps65217
 28:       1785      INTC  12  edma
 30:          1      INTC  14  edma_error
 33:        274      INTC  17  47400000.dma-controller
 34:     418377      INTC  18  musb-hdrc.0.auto
 35:          1      INTC  19  musb-hdrc.1.auto
 36:          0      INTC  20  4a300000.pruss
 37:          0      INTC  21  4a300000.pruss
 38:          0      INTC  22  4a300000.pruss
 39:          0      INTC  23  4a300000.pruss
 40:          0      INTC  24  4a300000.pruss
 41:          0      INTC  25  4a300000.pruss
 42:          0      INTC  26  4a300000.pruss
 43:          0      INTC  27  4a300000.pruss
 44:       6153      INTC  28  mmc1
 56:          0      INTC  40  4a100000.ethernet
 57:          0      INTC  41  4a100000.ethernet
 58:          0      INTC  42  4a100000.ethernet
 59:          0      INTC  43  4a100000.ethernet
 80:          0      INTC  64  mmc0
 84:      26851      INTC  68  gp_timer
 86:       1103      INTC  70  44e0b000.i2c
 88:         10      INTC  72  serial
 91:          0      INTC  75  rtc0
 92:          1      INTC  76  rtc0
 93:          0      INTC  77  wkup_m3
 94:          1      INTC  78  wkup_m3_txev
125:          0      INTC 109  53100000.sham
127:          0      INTC 111  48310000.rng
150:          0  44e07000.gpio   6  mmc0
**171:          0  44e07000.gpio  27  adis16209**
IPI0:          0  CPU wakeup interrupts
IPI1:          0  Timer broadcast interrupts
IPI2:          0  Rescheduling interrupts
IPI3:          0  Function call interrupts
IPI4:          0  Single function call interrupts
IPI5:          0  CPU stop interrupts
IPI6:          0  IRQ work interrupts
IPI7:          0  completion interrupts
Err:          0

Can you please confirm if I need to enable buffered operation, by writing to some registers (not sure which ones) on the device?

Thank you.

pcercuei, 2015-04-10:
Error 13 means "permission denied". You have to run the application with super-user rights.

Alternatively, if you have the IIO daemon ```iiod``` running on your target, you can create a network context with the loopback IP address ```127.0.0.1``` and that won't require super-user rights.

alphasilver, 2015-04-10:
Thanks, I run the application with sudo and iio_device_create_buffer now succeeds.

Despite this, I am getting another error in iio_buffer_refill. I guess this means connection timed out. Why could this be happening?

device : adis16209
 enabled ch0 : supply(input)
 enabled ch1 : accel_x(input)
 enabled ch2 : accel_y(input)
 enabled ch3 : voltage1(input)
 enabled ch4 : temp0(input)
 enabled ch5 : incli_x(input)
 enabled ch6 : incli_y(input)
 enabled ch7 : rot_x(input)
 enabled ch8 : timestamp(input)
WARNING: High-speed mode not enabled
Unable to refill buffer
ret: -110

pcercuei, 2015-04-10:
Err=-110 is a timeout, yes. Most likely the driver times out while waiting for the interrupt. You can verify that the interrupt fired in ```/proc/interrupts```.

alphasilver, 2015-04-10:
I'm doing cat /proc/interrupts before and after running the application, but I get always:
171:          0  44e07000.gpio  27  adis16209
If the interrupt fired does anything change in there?

How should I proceed to overcome this error?
Might this be because of wrong device register configuration? (I'm using all the defaults of ADIS16209).

alphasilver, 2015-04-10:
I was able to verify that the interrupt fired:
171:          1  44e07000.gpio  27  adis16209
272:          0  adis16209-dev0  adis16209_consumer0

What could be the cause of the timeout?
Might this be because of wrong device register configuration? (I'm using all the defaults of ADIS16209).

pcercuei, 2015-04-15:
What is the second line? The '0' means that it didn't receive any interrupt.

alphasilver, 2015-04-15:
Hi Paul,
Thanks for getting back on this issue.

I'm doing **watch -n2 "cat /proc/interrupts"** on another ssh session, while I run the application with sudo.
The full listing is below. I don't see any "second line".

Every 2.0s: cat /proc/interrupts                        Sun Mar  1 21:32:38 2015

           CPU0
 23:          1      INTC   7  tps65217
 28:       1511      INTC  12  edma
 30:          3      INTC  14  edma_error
 33:        438      INTC  17  47400000.dma-controller
 34:      87035      INTC  18  musb-hdrc.0.auto
 35:          1      INTC  19  musb-hdrc.1.auto
 36:          0      INTC  20  4a300000.pruss
 37:          0      INTC  21  4a300000.pruss
 38:          0      INTC  22  4a300000.pruss
 39:          0      INTC  23  4a300000.pruss
 40:          0      INTC  24  4a300000.pruss
 41:          0      INTC  25  4a300000.pruss
 42:          0      INTC  26  4a300000.pruss
 43:          0      INTC  27  4a300000.pruss
 44:       4730      INTC  28  mmc1
 56:          0      INTC  40  4a100000.ethernet
 57:          0      INTC  41  4a100000.ethernet
 58:          0      INTC  42  4a100000.ethernet
 59:          0      INTC  43  4a100000.ethernet
 80:          0      INTC  64  mmc0
 84:       5391      INTC  68  gp_timer
 86:       1103      INTC  70  44e0b000.i2c
 88:         10      INTC  72  serial
 91:          0      INTC  75  rtc0
 92:          1      INTC  76  rtc0
 93:          0      INTC  77  wkup_m3
 94:          1      INTC  78  wkup_m3_txev
125:          0      INTC 109  53100000.sham
127:          0      INTC 111  48310000.rng
150:          0  44e07000.gpio   6  mmc0
**171:          0  44e07000.gpio  27  adis16209
272:          0  adis16209-dev0  adis16209_consumer0**
IPI0:          0  CPU wakeup interrupts
IPI1:          0  Timer broadcast interrupts
IPI2:          0  Rescheduling interrupts
IPI3:          0  Function call interrupts
IPI4:          0  Single function call interrupts
IPI5:          0  CPU stop interrupts
IPI6:          0  IRQ work interrupts
IPI7:          0  completion interrupts
Err:          0

pcercuei, 2015-04-15:
By "second line" I meant the one that starts with 272.
Anyway, you don't receive interrupts so that's the reason why you experience timeouts.

alphasilver, 2015-04-15:
Can you point out any possible reasons why this might be happening?

The interrupt seems to be well configured in the device tree. I don't see any conflicts. I there any other way to confirm if there is a problem with the interrupt?

The only thing left I have to check is the device register configuration. As you see in my source code, I didn't change any of the default settings of the ADIS16209. Could that be the reason?

pcercuei, 2015-04-17:
The timeout could also occur because the SPI communication does not work.
For instance, I see that you enabled pull-ups on all the SPI pins, you need to disable them.

alphasilver, 2015-04-17:
I've rebuilt the device tree with:
```c    spi0_pins: pinmux_spi0_pins {
        pinctrl-single,pins = <
            0x150 (PIN_INPUT_PULLDOWN | MUX_MODE0)    /* spi0_sclk.spi0_sclk */
            0x154 (PIN_INPUT_PULLDOWN | MUX_MODE0)    /* spi0_d0.spi0_d0 */
            0x158 (PIN_OUTPUT_PULLDOWN | MUX_MODE0)   /* spi0_d1.spi0_d1 */
            0x15c (PIN_OUTPUT_PULLDOWN | MUX_MODE0)   /* spi0_cs0.spi0_cs0 */
        >;
    };
```
but the result is the same: Unable to refill buffer, ret: -110.

pcercuei, 2015-04-17:
Well, enabling pull-downs won't work either...

alphasilver, 2015-04-17:
I was looking at the timing diagram of the ADIS16209 and enabling pull downs seemed correct.

I've rebuilt the device tree with PIN_INPUT / PIN_OUTPUT instead of PIN_INPUT_PULLDOWN / PIN_OUTPUT_PULLDOWN but the result when running the application is the same. 

What do you mean exactly I should do?

pcercuei, 2015-04-20:
If you read 0x4A with iio_device_reg_read, what value do you obtain? (be sure to enable debugfs first)
It should give you 0x3F51, that's the product identification code.
If you get a different value or an error then there is no SPI communication.

alphasilver, 2015-04-21:
You are right Paul, I tried to read PROD_ID as you suggested and obtained an error return value (-2 No such file or directory). So I have to agree I don't have SPI communication.

Actually this caught me by surprise because I thought that once the device is present in /dev/iio:device* and in /sys/bus/iio/devices/ it would communicate. 

Right now my problem is that everything seems to be correct and I can't find the cause for SPI communication not working. Maybe you can suggest something I can try.

The ADIS16209 ADIS1620x/PCB is connected to the Beaglebone black as follows:
BBB P9_01 (GND) -> J1_07
BBB P9_04 (DC_3.3V) -> J1_11 (VDD)
BBB P9_17 (spi0_cs0 : 0x15c) -> J1_03 (CS)
BBB P9_18 (spi0_d1 : 0x158) -> J1_06 (DIN)
BBB P9_21 (spi0_d0 : 0x154) -> J1_04 (DOUT)
BBB P9_22 (spi0_sclk : 0x150) -> J1_02 (SCLK)
BBB P9_12 (GPIO1_28 : 0x78) -> J2_12 (DIO1)

The device-tree on the BBB is as follows: 
```c
        spi0_pins: pinmux_spi0_pins {
                pinctrl-single,pins = <
                        0x150 (PIN_INPUT | MUX_MODE0)   /* spi0_sclk.spi0_sclk */
                        0x154 (PIN_INPUT | MUX_MODE0)   /* spi0_d0.spi0_d0 */
                        0x158 (PIN_OUTPUT | MUX_MODE0)  /* spi0_d1.spi0_d1 */
                        0x15c (PIN_OUTPUT | MUX_MODE0)  /* spi0_cs0.spi0_cs0 */
                >;
        };

&spi0 {
        pinctrl-names = "default";
        pinctrl-0 = <&spi0_pins>;
        status = "okay";

        spidev0: spi@0 {
                compatible = "adi,adis16209";
                reg = <0>;
                spi-max-frequency = <16000000>;
                spi-cpha;

                interrupt-parent = <&gpio1>;
                interrupts = <28 IRQ_TYPE_EDGE_RISING>; /* P9_12: gpio1[28] */
        };

        spidev1: spi@1 {
                compatible = "spidev";
                reg = <1>;
                spi-max-frequency = <16000000>;
        };
};
 ```
Thank you very much for your help.

pcercuei, 2015-04-21:
Actually the "no such file or directory" most probaby means that you don't have the "direct_reg_access" file in debugfs (or that you didn't mount debugfs).

I just had a look at the driver, it does test the presence of the chip so if the driver doesn't issue an error then the SPI communication is fine.

alphasilver, 2015-04-21:
I have in /etc/fstab: **debugfs  /sys/kernel/debug  debugfs  defaults  0  0**, so the debugfs should be always mounted. The /sys/kernel/debug is populated, but I can't find the direct_reg_access file you mentioned.

```
root@beaglebone:/sys/kernel/debug# ls
asoc	       extfrag	  memblock		 omap_mux    sched_features
bdi	       f2fs	  mmc0			 pinctrl     sleep_time
binder	       frontswap  mmc1			 pm_debug    suspend_stats
clk	       gpio	  musb-hdrc.0.auto	 pwm	     sync
dma_buf        hid	  musb-hdrc.0.auto.dsps  regmap      tracing
dri	       iio	  musb-hdrc.1.auto	 regulator   usb
dynamic_debug  kprobes	  musb-hdrc.1.auto.dsps  remoteproc  wakeup_sources
```
When the system boots I get a few of these messages, but no errors:
```
dmesg
...
[    2.941044] adis16209 spi1.0: setup: speed 12000000, sample trailing edge, clk normal
[    2.941103] adis16209 spi1.0: setup: speed 12000000, sample trailing edge, clk normal
[    2.941150] adis16209 spi1.0: setup: speed 12000000, sample trailing edge, clk normal
...
```

So, right now I still have no clues about what could be wrong... Any suggestions?

pcercuei, 2015-04-21:
The "direct_reg_access" file should be somewhere inside the "iio" directory. But the error that you get before (-2) seems to indicate that this file is not present (I don't know when it is and when it is not).

It still looks like an interrupt problem to me. You could just put some power to the GPIO that you normally connect to the ADIS chip, just to see if you get interrupts in ```/proc/interrupts```.

alphasilver, 2015-04-21:
I will try to test the interrupt as you suggest and get back to you soon.

I just checked /sys/kernel/debug/iio and there is nothing inside the directory. Why could this be happening?  Maybe there's something that need to be activated in the kernel config.
I noticed that there is an option to "Expose hardware/virtual IRQ mapping via debugfs" in General setup > IRQ subsystem. Should I activate that?

alphasilver, 2015-04-23:
Hi Paul,
I have performed a few tests as you requested. 

1. First, with the ADIS16209 disconnected, I tested the SPI bus by connecting together the spi0_d1 and spi0_d0, and successfully transmitted a block of data using the Linux user space SPI API with ioctl requests.
2. Then reconnected the ADIS16209 and tested the interrupts in the IRQ pin gpio0[27]. I meant to use Analog Discovery to generate a digital signal with 1 Hz rate, but as soon as I connected the AD to the IRQ pin I started getting interrupts, even before I start running the pattern generator!  Surprisingly the interrupts keep firing even when the signal is a constant zero (I confirmed this with the oscilloscope). If then I run the application the line 272 below appears and the app keeps running for a long time without error (much longer that it would be supposed to).
```
150:          0  44e07000.gpio   6  mmc0
171:      19248  44e07000.gpio  27  adis16209
272:       2174  adis16209-dev0  adis16209_consumer0
IPI0:          0  CPU wakeup interrupts
```

For me the questions about what could be wrong remain the same:
* Why the directory /sys/kernel/debug/iio isn't populated? Is it a problem of kernel configuration? Is this why I can't read or write the device registers?
* Why do you say it seems an interrupt problem? As far as I understand the interrupt is required only because I need buffered operation - reading and writing to the device registers should work.
* Reading the ADIS16209 datasheet it seems that buffered operation is not enabled by default - MSC_CTRL = 0x00 by default, so bit 2 means data-ready disabled. The problem is that I can't write to the register.
* Is production code based on libiio supposed to work always with debugfs enabled?

Can you please suggest something I could do? Thank you for your help.

pcercuei, 2015-04-23:
I had a quick look at the datasheet. Your SPI clock is way too fast, it should be 1 MHz maximum, and you configure it for 16 MHz.

For your other questions:
* I suppose that you are not using our Linux tree (https://github.com/analogdevicesinc/linux/). Some parts of the IIO subsystem are still not upstream; if you can, try to build our kernel (branch xcomm_zynq) for your BBB. This might explain why you don't have the debugfs files, for instance.
* Reading and writing to device registers appear to be working, otherwise the driver wouldn't probe at all (still, try to lower the SPI clock).
* You should not try to write to MSC_CTRL, the driver will do it when you enable the buffer.
* Libiio works fine with and without debugfs enabled.

pcercuei, 2015-04-27:
Ok, at least I can reproduce it - with an ADIS16334.
I will debug the issue and get back to you.

alphasilver, 2015-04-27:
I have reduced the SPI clock to 1MHz as you suggested but the result is always the same.

At least part of the problem seems to be that the interrupt is not being fired. (Also for me it is surprising that when I connected the Analog Discovery to the IRQ pin I started getting interrupts, even before I start running the pattern generator). I'll try to investigate if this could be a kernel issue (if it was I wouldn't get interrupts with the Analog Discovery, would I?), and will try other kernels for BBB by Robert Nelson. Right now I'm using the 3.14.35-ti-r55 kernel. To use the kernel from Analog Devices I guess I would loose a great number of patches for BBB, so that wouldn't be my first choice. 

In the mean time I have connected DIO1 of the ADIS16209 to an oscilloscope and got a bit surprised of the signal shape, see the next picture. (maybe I'm doing something wrong) The signal shape is always the same (I don't see any difference due to driver configuration during or after running the application). I'm connecting directly DIO1 to the BBB gpio pin, do I need to put a resistor in between?
[image]

I'm glad if you can debug this issue. I have reviewed everything over and over and don't know what I should do next.
[image]
[image]

Thank you for your help.

lclausen-adi, 2015-04-27:
Hi,

That signal looks very wrong. Your signal level should not exceed 3.3V. Can you double check that all the ground signals are properly connected? BBB to ADIS16209 to logic analyzer.

alphasilver, 2015-04-27:
Should I connect J1 pins 7 and 8 together to BBB GND, or just pin7 is enough? Are all my connections correct?

The signal looked so odd that I repeated I have repeated the measurement a number of times with the same result. I still suspect that ground might not be properly connected as the access below the pin headers is not easy. I'm welding a new cable so that I can check that again and get back to you.

lclausen-adi, 2015-04-27:
The GND pins on the ADIS breakout board are all internally connected, so connecting one of them to the BBB should be enough. But make sure that the GND signal of your oscilloscope is also connected to the circuit.

The signals look good. RST is not connected, but according to the datasheet it has an internal pull-up, but you could still try to connect it to 3.3V and see if that makes a difference.

alphasilver, 2015-04-28:
Here are the results of my measurements:

**BBB,GND - P9_12** (ADIS16209 connected, P9_12 connected to J2,12-DIO1)
[image]

**BBB,GND - J2,12-DIO1** (J2 pin header disconnected from the BBB = P9_12 disconnected)
[image]

**BBB,GND - P9_12** (P9_12 disconnected from J2,12-DIO1)
[image]

**BBB,GND - P9_12** (P9_12 connected to DIO1+DC_3.3V connected to J1,1-RST)
[image]

pcercuei, 2015-04-30:
I pushed a fix in the kernel that **might** help you: https://github.com/analogdevicesinc/linux/commit/5e19efc5d3154569e3991e5b64f87afd36e677f8

Although if DIO1 gives you 10V then there's another problem somewhere... the chip works at 3.3V...

alphasilver, 2015-04-30:
Hi Paul,
Just one question: If I'm using the 16209, why do you say the fix to the driver of 16400 might help? I was taking a look at *drivers/staging/iio/accel/adis16209.h* and *drivers/staging/iio/accel/adis16209_core.c* and I don't see any includes for the files you updated.

The 10V were measured between the GND on BBB and DIO1 disconnected from the BBB. I don't see any logical explanation how it can give 10V. Am I doing something wrong? All the wiring on J1 seems correct. How should I proceed to debug this issue? Could the sensor be damaged?

Thank you.

pcercuei, 2015-05-19:
Hi, Sorry for the (very) late reply.

Yes, I got confused, the fix I pushed was only for adis16400.

Are you 100% certain that you don't power the chip with 12V? Or something like that. I am not an electrical engineer, so I'm afraid I can't really help you with that. I suggest that you ask in our engineer zone: http://ez.analog.com

pcercuei, 2015-06-23:
I will close this issue - last message was almost 2 months ago, and the problem is most likely not related to libiio.

closed 2015-06-23 by pcercuei

skfelix, 2018-02-28:
try installing the last version of libiio from the source code available on the libiio's github. The file is a .tar.gz file. When compiling, you will need to type like this:
g++ -std=gnu++11 -I/root/libiio-0.14/install/include -L/root/libiio-0.14/install/lib code.cpp -o code -liio
The paths might change according to how you have installed libiio. I recommend you to install as described next:
1 - install the packages listed in https://wiki.analog.com/resources/tools-software/linux-software/libiio
2 - download the source code in https://github.com/analogdevicesinc/libiio/releases
3 - unzip
4 - 
# mkdir install
# cd libiio-0.1x/
# cmake -DCMAKE_INSTALL-PREFIX=/<your-path>/libiio-0.1x/install
# make install
In case you have problems with the install, delete the install/ folder