analogdevicesinc/libiio #11: local_get_buffer() Valgrind error

njpillitteri, 2015-05-15:
I'm trying to debug what I think it stack corruption in my program, and I came across this. I'm not sure if it's related to my problem.

==865== Thread 2:
==865== Syscall param ioctl(generic) points to uninitialised byte(s)
==865==    at 0x4B8B76C: ioctl (syscall-template.S:81)
==865==    by 0x487C2C7: local_get_buffer (in /usr/lib/libiio.so.0.5)
==865==  Address 0x57f6d38 is on thread 2's stack
==865==

lclausen-adi, 2015-05-18:
Hi,

What happens here is that an uninitialized memory region is passed to the ioctl function, which is OK in this case since the kernel will initialize the memory region. It's just that valgrind doesn't know about these specific semantics of the IIO ioctls.

But anyway we already fixed it a while ago by fully initializing the struct see commit cb8104b19798ef569d3adbad6123a4876be1c5f4

njpillitteri, 2015-05-18:
I was up to date and had that commit and still getting a Valgrind error. I pulled a commit that fixed it by memsetting a struct to 0s.

pcercuei, 2015-05-19:
I merged your pull request, so I'll close this issue now.
Thanks!

closed 2015-05-19 by pcercuei