analogdevicesinc/libiio #16: Enhance buffer API

rofferom, 2015-06-22:
I have some requests about buffer API. Actually I have some needs, but I would like to have a feedback before starting my development, mainly to know if the ideas themself are correct for libiio, and to have some clues about the expected implementation. The goal is to provide a pull request that has a sense...

My goal is to fetch samples from three differents sensors that are running at different rates, in an unique thread. I would like to :
* Make a blocking read on my faster sensor (no problem here, it's already available in libiio API)
* Read the other sensors. If no data are available, get EAGAIN. There is already a iio_context_set_timeout() but it is global to the context, not locally defined for a buffer

Furthermore, I would like to have a way to control the size of the buffer on kernel side. From what I understand, libiio configures the buffer at the exact size of the read done by iio_buffer_refill(), and this size is configured in iio_device_create_buffer(). Actually I would like to use kernel buffer to store some extra data for me, in case I'm a little bit late in my processing, so a little tuning of the kernel buffer size would help me a lot.

To conclude, I would like to add these functionalities in the buffer API :
* Support of non-blocking reads
* Kernel buffer size configuration

An extra feature is to export the filedescriptor of the buffer to read (or return -1 if no pollable filedescriptor is available), to do the poll() myself in my acquisition loop.

pcercuei, 2015-06-24:
Hi,

First, about the kernel size configuration. Right now the high-speed interface (mmap + ioctl based) uses 4 kernel buffers (hard-coded value). The low-speed interface (stdio) uses only one. For now I will increase the number of buffers to 4 as well as a quick fix, but it would be great to have the number of buffers configurable through an API function so that you can adjust the latency.

The question about non-blocking reads is a bit tricky, if you consider that the applications using libiio must run the exact same independently of the backend used (local or network) and the OS.

For instance, exporting the file descriptor is not a solution, as you would get a different behaviour whether you are running with the local backend or the network backend; with the latter, you have to ask the server for data first, so polling the file descriptor won't work.

The solution might be to add API functions "iio_buffer_refill_nonblock" and "iio_buffer_push_nonblock" (alternatively, "iio_buffer_set_nonblock" and have "iio_buffer_refill" and "iio_buffer_push" return -EAGAIN). Then, you can handle this differently in the backends so that it works homogeneously across all backends.

rofferom, 2015-06-24:
Tank for the commit. For the configuration, I'm trying to guess that kind of API function can be added. According to the constant NB_BLOCKS, should it be iio_device_set_nb_blocks(), that should be called before iio_device_create_buffer() (as this function configures the kernel buffer size) ?

About the non-blocking API, I suggest to use a iio_buffer_set_nonblock() with iio_buffer_refill() and iio_buffer_push() that returns -EGAIN. This behaviour is the same that non-blocking sockets.

I understand the problem of polling in case of a network backend. I don't know your policy about the way to handle the different types of backends, but I can see that some operation in public API can return -ENOSYS if the backend doesn't provide an implementation of the function. Would it be possible to consider that only local backend provides a function that gives a pollable fd (I'm not sure how this can be implemented for high_speed devices) ?

rofferom, 2015-06-24:
I have worked on a first version of the three API enhancement, to make some tests on my side. I have pushed this on my repo. Note this is a draft, this is not fully integrated on my side, but the first tests seem to work.

If you want to have a preview of the API and the implementation :
* Add iio_buffer_set_blocking_mode() API : rofferom/libiio@d30c021fa435ca0bc95135cc50cef2e2d9d87e12
* Add iio_buffer_get_fd() API : rofferom/libiio@33a19d30ca38389203219517b2166ec86b03c16b
* Add iio_device_set_nb_blocks() API : rofferom/libiio@4285301a02b4fe368887a16433aa1758c1899786

pcercuei, 2015-06-26:
Hi,
Hi,

I had a quick look and I have some comments:
 * The commit that adds "iio_buffer_set_blocking_mode( )" changes everything to avoid the FILE API. It makes the patch very big and I don't understand the reason.
 * For the second commit I'd prefer the API function to be called "iio_buffer_get_poll_fd", so that it's clear that it's not meant for IO access. Besides, you do a check on !dev_is_high_speed, but you can actually poll the file descriptor even for drivers that use the high-speed interface. Also, the doxygen documentation should be enhanced a bit ("fd" -> "file descriptor", and the function doesn't return -1 on error but a negative errno code).
 * The "iio_device_set_nb_blocks( )" commit gets the idea but it can be improved. Since it is not possible to change the number of blocks while the device is opened, I would just make the "local_set_nb_blocks" function set pdata->nb_blocks, and allocate pdata->blocks and pdata->addrs inside "enable_high_speed( )" (since those are only useful for high-speed devices) and free them in local_close. Also I would prefer the API function to be called "iio_device_set_kernel_buffers_count" since it might not be clear to everybody what a "block" is.

rofferom, 2015-06-26:
Thank for you feedback.

* Actually I think it is unnecessary to use FILE API as we want to have a non blocking mode. In this case the buffering of the FILE API is something that should be avoided. This is just for this reason that I have changed the API used internally. But in any case, I think I should have suggested this change in a separate commit.
* Ok, I wasn't sure about the name of the API, I will update this
* Same for the previous point !

I will send a pull request when the FILE API point will be resolved, I will be easier to clarify the details of the implementation.

closed 2015-07-01 by rofferom