analogdevicesinc/libiio #23: Function read_all (local.c) return -5 when data is available

jcsistemas2001, 2015-12-18:
In a acquisition system with 36 bytes by sample, a buffer is created [iio_device_create_buffer(dev, 1000, false)], and trying to fetch 2 samples, the function iio_buffer_refill return -5. 

After debugging, the function read_all (at local.c) return -5 in the last condition because the function read the samples partially (not all bytes defined in "len").

    static ssize_t read_all(void *dst, size_t len, int fd)
    {
	uintptr_t ptr = (uintptr_t) dst;
	ssize_t readsize;
	int ret;
	while (len > 0) {
		do {
			ret = read(fd, (void *) ptr, len);
		} while (ret == -1 && errno == EINTR);

		if (ret == -1) {
			ret = -errno;
			break;
		} else if (ret == 0) {
			ret = -EIO;
			break;
		}

		ptr += ret;
		len -= ret;
	}
	readsize = (ssize_t)(ptr - (uintptr_t) dst);
	if ((ret > 0 || ret == -EAGAIN) && (readsize > 0)) 
		return readsize;
	else 
		return ret;
    }

If the samples count (in my case 2) are not in the driver buffer, the library discard any sample acquired and return -5.  

I suggest to two changes:

- Not change "ret" to "-EIO" when read return 0
- Compare "ret" with ">=" in last condition

Below the suggestion in the code:

    static ssize_t read_all(void *dst, size_t len, int fd)
    {
	uintptr_t ptr = (uintptr_t) dst;
	ssize_t readsize;
	int ret;
	while (len > 0) {
		do {
			ret = read(fd, (void *) ptr, len);
		} while (ret == -1 && errno == EINTR);
		if (ret == -1) {
			ret = -errno;
			break;
		} else if (ret == 0) {
			// >>>> ret = -EIO;  <<<<
			break;
		}
		ptr += ret;
		len -= ret;
	}
	readsize = (ssize_t)(ptr - (uintptr_t) dst);
	// >>>> if ((ret > 0 || ret == -EAGAIN) && (readsize > 0)) <<<<
	if ((ret >= 0 || ret == -EAGAIN) && (readsize > 0)) 
		return readsize;
	else
		return ret;
    }

pcercuei, 2015-12-18:
Sorry, I have to disagree. The read( ) call, by the POSIX definition, is guaranteed to read at least one byte on success. A return value of zero means that we reached the end of the file, or that the pipe was closed, which in that case is a real I/O error.
So the error here seems to be that the kernel driver's read( ) backend function returns 0 when it should not.

jcsistemas2001, 2015-12-18:
Thanks for your reply. I underand your statement about to follow the standard. However, it's not clear to me, that the function must return 0 when the pipe is closed, I think must return -1 with ernno (http://pubs.opengroup.org/onlinepubs/009695399/functions/read.html).
In any case, may I know your suggestion about the value that must be returned by the driver (kfifo) to inform the application that there is not data available?
It's odd because nobody has noted data loss (based on iio_readdev example).

pcercuei, 2015-12-22:
It's in the link you pasted: "If no process has the pipe open for writing, read() shall return 0 to indicate end-of-file.". The only other case where 0 can be returned is when you try to read 0 bytes.

The driver should block until there is data available. Or, in the case where the device node has been opened with O_NONBLOCK, it can make the read( ) call return -1 with an error code of EAGAIN.

In libiio, you can force the device node to be opened in non-blocking mode using iio_buffer_set_blocking_mode( ), see http://analogdevicesinc.github.io/libiio/group__Buffer.html#ga8306f2992b49b99b6456907b65ece5aa.

jcsistemas2001, 2015-12-28:
Hi Paul, thanks for your suggestion and explanation.

closed 2015-12-28 by jcsistemas2001