analogdevicesinc/libiio #24: Network versus local context

jcsistemas2001, 2016-02-21:
Hi Paul,

I'm running a system with an ADC conversion at 1000Hz and noted that sometimes using a network context, there is a package lost (127.0.0.1). I describe below how the drop packages are measured.

The application is checking the timestamp (setup with monotonic clock) to validate there is not data lost in the SPI interruption (currently around 30% CPU usage). The same application calculate the time between samples to diagnostic data lost. Could be another way to perform this checking at user space level using libiio?

At the moment, the **samples_count** is set to be 0.1 sec (100 samples) when the buffer is created  _iio_device_create_buffer (pDev, 100, false)_. 

Sometimes (around minutes) the time between samples is 100ms (almost exact). Ir seems there is a package lost anywhere. Assuming this setup, may you give a hint where the data is lost and how to avoid it? Any other test setup to evaluated this issue? 

BTW, I used this comm channel for previous experience. Let me know if there is another more convenient (I don't want to register an issue here without to be sure about it).

Thanks

pcercuei, 2016-02-22:
Hi,

The proper comm channel is [our engineering zone](https://ez.analog.com/community/linux-device-drivers/linux-software-drivers).

Internally, in the local context libiio will create 4 buffers (by default), so having a timestamp of 100ms between two samples does not mean that data was lost.

What I would suggest to detect dropped samples: capture a ~250 Hz waveform, and look at the FFT of the captured signal (for instance using the [IIO Oscilloscope](https://github.com/analogdevicesinc/iio-oscilloscope); then the discontinuities of the stream become really visible.

-Paul

jcsistemas2001, 2016-02-22:
Hi Paul,

Thanks for the suggestion. 

Inundertand that a time stamp of 100ms doesn't mean that data is lost, but if the difference between two continue time stamps is 100ms, there are some sample lost in somewhere, because the system is acquiring with 1ms between samples. In this case, any idea where the data is lost?

I agree about to test with a pure sinoide and check the FFT. However, could be a tedious task to check the FFT during long periods of time or implement an approach to detect lower or higher frequency automatically. Any other approach?

Thanks

Julio

rgetz, 2016-02-22:
Depending on the converter - many have a test most which generate a ramp, or PN sequence - it should be easy to (a) turn the test mode on (b) capture samples on the host (c) check the test mode in the host to make sure buffers aren't being lost...

You should be able to run that as long as you want.

-Robin

jcsistemas2001, 2016-02-23:
Hi Robin,

Thanks you to add this approach evaluation. 

In the beginning, I used this approach, generating the PN sequence in the iio driver interrupt. I will recheck again (in test mode). 

In operation mode, I will need another approach. In this case, I use the timesstamp, that could solve both modes. 

In my case, the system cannot lose samples without notification to user space.

I will post some test results later. Right now, the system is acquiring some signals.

Thanks

Julio

jcsistemas2001, 2016-02-23:
Hi Paul, Robin,

Two additional tests were performed with the following conditions:

1. Network context
2. Local context 

Both testing were perform during some hours with the following conditions:

- **Device**: send samples @ 4000Hz (sample time or period: 250us)
- **Libiio config**: samples_count: 400 samples [iio_device_create_buffer (pDev, 400, false)]
- **Log file** including "date/time" sample number (from zero), sample time (ns) and timestamp (ns) when the following condition fail "200us > sample time > 300us".
- The CPU is a iMX Solo with a maximum load around 50%

The results show no error using the local context. However, show errors during the network context test. More details below:

- Log (Network): https://justpaste.it/rmlh
- Log (Local): https://justpaste.it/rmlp

------
Error processing latency: "Tue Feb 23 17:16:41 2016" 33728892;181000;-1973175853
Error processing latency: "Tue Feb 23 17:25:12 2016" 35773676;300667;-1874502683
Error processing latency: "Tue Feb 23 17:26:49 2016" 36162401;100251000;918126150 **<< issue**
Error processing latency: "Tue Feb 23 17:27:16 2016" 36268912;301667;1776317044
Error processing latency: "Tue Feb 23 17:27:16 2016" 36268913;198333;1776515377
------

Thanks again for your help!

rgetz, 2016-02-23:
Julio:

So we can try to replicate your results - what converter are you using?

I did this awhile ago with the AD9361, and didn't see any issues (PN sequence generated inside the converter, connected to a Xilinx Zynq, streaming to an x86 host over Gig Ethernet). 

I can dig out the test code, and commit it somewhere so you can have a look. 

- Robin

jcsistemas2001, 2016-02-24:
Hi Robin,

That's sound great (replicate the results). However, my system is something custom (no supported ADC).   
Maybe, I could replicate the results using a simple driver (without device). I will prepare this environment and let you guys know the results. In case I can replicate the results, I send you the code to test at your place.

I will also compare the code with the AD9361 driver, maybe there is something could be missing at my side. 

About the test code you mention, sure will be great to see if there are something that I'm missing.

Thanks

Julio

lclausen-adi, 2016-02-24:
Hi,

Just to add what already has been said. It is always possible that due to back-pressure from the network transport that you loose data at the source. E.g. if the network is not capable of transmitting the data at the very moment your buffer might overflow and you loose data. Another source of disruption might be that some driver disables interrupts for an extended period. One thing that can help is e.g. larger buffers to compensate for jitter. You'll have to dive into the issue yourself and see where at the lowest levels the overflows occur to properly analyse and understand the issue.

- Lars

jcsistemas2001, 2016-02-27:
Hi Lars,

Thanks for your comments and suggestions.

Currently, the system is using a RT patch to control the latency. In fact, without that, was impossible to acquire all the SPI samples (as discussed some weeks ago at IIO list). I already perform some cyclictest with success (around 125us max).

In any case, as you All suggested, I will try to dive into the issue by myself.

For now, I will close the issue here and continue later when I have more results, at the engineering zone (as suggest Paul).

Regards

closed 2016-02-27 by jcsistemas2001