analogdevicesinc/libiio #25: Handling channels that have no modifier or are not indexed

gburca, 2016-03-04:
iio_channel_is_scan_element() returns false for channels without a modifier and that are not indexed, even though the kernel creates an entry in the scan_elements directory for them. Is that correct?

For example, a temperature sensor would have a single channel (so there's no need to index it), and it wouldn't need a modifier either. It would show up as "in_temp_raw" in the device directory and have "in_temp_{en|index|type}" files in the scan_elements directory.

lclausen-adi, 2016-03-05:
Hi,

It's not intentional. Please send a patch.

Thanks,
- Lars

abrauchli, 2017-11-22:
Potentially fixed by https://github.com/analogdevicesinc/libiio/pull/139

gwendalcr, 2019-05-10:
#139  has been reverted by fb5fa69.  #273 addresses the problem, but how do I ensure it does not add a regression?

gwendalcr, 2020-02-14:
#366 fixes the issue.

rgetz, 2020-03-31:
Ok - since it's fixed, I will close. Thanks

closed 2020-03-31 by rgetz

gwendalcr, 2020-11-19:
I was wrong, it is not fixed.
For simple attribute, like in_illuminance, found on driver like acpi_als, is_channel() returns false and iio_info does not report in_illuminance as a channel, iio_readdev can not read it.

With a trivial path (forcing is_channel("in_illuminance_..", true) to return true, similar to commit 4c6baf5e ("Local backend: Properly detect "timestamp" as a channel"), iio_genxml present acpi_als illuminance as channel:

from:
  <device id="iio:device0" name="acpi-als">
    <channel id="illuminance" type="input">
      <attribute name="raw" filename="in_illuminance_raw"/>
    </channel>
...

to:
  <device id="iio:device0" name="acpi-als">
    <channel id="illuminance" type="input">
      <scan-element index="0" format="le:S32/32&gt;&gt;0"/>
      <attribute name="raw" filename="in_illuminance_raw"/>
    </channel>
 
Since #273 is too generic, should we look at scan_elements/ directory to identify channels instead of the attributes at the root of iio:deviceX?