analogdevicesinc/libiio #27: CMake Cross Compiling

zmb3, 2016-03-17:
I had to make a few changes to the CMake files in order to get libiio to compile with a GCC Linaro ARM toolchain.

1) Make the `iio` target's link libraries public.

```
-target_link_libraries(iio LINK_PRIVATE ${LIBS_TO_LINK})
+target_link_libraries(iio PUBLIC ${LIBS_TO_LINK})
```
Without this change, `iiod` links to libiio, but doesn't link libiio's dependencies, so you get errors like the following:

```
$ make
Scanning dependencies of target iio
[  4%] Building C object CMakeFiles/iio.dir/channel.c.o
[  8%] Building C object CMakeFiles/iio.dir/device.c.o
[ 12%] Building C object CMakeFiles/iio.dir/context.c.o
[ 16%] Building C object CMakeFiles/iio.dir/buffer.c.o
[ 20%] Building C object CMakeFiles/iio.dir/utilities.c.o
[ 24%] Building C object CMakeFiles/iio.dir/local.c.o
[ 28%] Building C object CMakeFiles/iio.dir/network.c.o
[ 32%] Building C object CMakeFiles/iio.dir/xml.c.o
[ 36%] Building C object CMakeFiles/iio.dir/iiod-client.c.o
[ 40%] Linking C shared library libiio.so
[ 40%] Built target iio
[ 44%] [FLEX][lexer] Building scanner with flex 2.5.39
[ 48%] [BISON][parser] Building parser with bison 3.0.2
Scanning dependencies of target iiod
[ 52%] Building C object iiod/CMakeFiles/iiod.dir/iiod.c.o
[ 56%] Building C object iiod/CMakeFiles/iiod.dir/ops.c.o
[ 60%] Building C object iiod/CMakeFiles/iiod.dir/parser.c.o
[ 64%] Building C object iiod/CMakeFiles/iiod.dir/lexer.c.o
[ 68%] Linking C executable iiod
arm-linux-gnueabihf/bin/ld: warning: libxml2.so.2, needed by ../libiio.so.0.6, not found (try using -rpath or -rpath-link)
../libiio.so.0.6: undefined reference to `xmlFreeDoc@LIBXML2_2.4.30'
../libiio.so.0.6: undefined reference to `iio_mutex_create'
../libiio.so.0.6: undefined reference to `xmlReadMemory@LIBXML2_2.6.0'
```

2) Use `find_package(Threads)` instead of `find_library(PTHREAD_LIBRARIES pthread)`:

```
if (NEED_THREADS)
   if (NOT_WIN32)
-		find_library(PTHREAD_LIBRARIES pthread)
-		if (PTHREAD_LIBRARIES)
-			set(LIBS_TO_LINK ${LIBS_TO_LINK} ${PTHREAD_LIBRARIES})
-			set(LIBIIO_CFILES ${LIBIIO_CFILES} lock.c)

+		find_package(Threads)
+		if (CMAKE_USE_PTHREADS_INIT)
+			set(LIBS_TO_LINK ${LIBS_TO_LINK} ${CMAKE_THREAD_LIBS_INIT})
+			set(LIBIIO_CFILES ${LIBIIO_CFILES} lock.c)
		else()
			message(SEND_ERROR "The pthread library is a requirement for the selected backends")
		endif()
```

(And then also update iiod/CMakeLists.txt to link `${CMAKE_THREAD_LIBS_INIT}` instead of `${PTHREAD_LIBRARIES}`).

pcercuei, 2016-03-17:
Hi,

About 1): We link privately the libiio library on purpose. See commit d94994a.
What's your CMake configuration?

About 2): Could you send a pull request for this? (don't forget to sign-off your commit).

commodo, 2018-05-07:
@zmb3 
is this still an issue ?
or, any feedback ?

if no objections, or no response [ within a week ], i say we close this issue ;
feel free to re-open it later if the case

thanks
Alex

lclausen-adi, 2018-05-07:
2) sounds like a sensible change.

zmb3, 2018-05-07:
Sorry, I’m not sure if this is still an issue - I had totally forgotten about this. I haven’t used libiio or CMake since I wrote this issue so I’m having trouble recalling the details.

pcercuei, 2018-05-07:
I think 2) was a problem too; because it might not give you pthreads (e.g. if C11 threads are available).

rgetz, 2020-01-13:
I think people are cross compiling (based on #330) so closing this.

-Robin

closed 2020-01-13 by rgetz