analogdevicesinc/libiio #31: unable to create IIO buffer

mustaphahaydar, 2016-06-16:
Hi,
I try to connect the matlab windows with linux using the library libiio, I use in my project a zedboard execute Linux connected to AD9361.

At the beginning he show that he "cannot connect to iiod" but when I execute the iiod manually (by writing iiod in the terminal), the connection with iiod establishes. Consequently when he started to upload data, it appears in the terminal:

- new Client connected from 10.1.25.142
- new Client connected from 10.1.25.142
- WARNING: Hight-speed mode not enabled
- ERROR: Unalbe to create buffer
- Client exited

and the  matlab crashes.
What Could Be the Reason That ERROR?

Thanks for your help in advance!
Sorry for my englsih

lclausen-adi, 2016-06-16:
Hi,

How do you invoke the capture on the matlab side?

- Lars

mustaphahaydar, 2016-06-16:
Hi
I used this example in matlab for data stream, and class iio_sys_obj_matlab and class libiio_if

s = iio_sys_obj_matlab; % Constructor
s.ip_address = '10.1.25.142';
s.dev_name = 'ad9361';
s.in_ch_no = 2;
s.out_ch_no = 2;
s.in_ch_size = 8192;
s.out_ch_size = 8192;

s = s.setupImpl();

input = cell(1, s.in_ch_no + length(s.iio_dev_cfg.cfg_ch));
Fs = 30.72e6;
Fc = 1e6;
t = 1/Fs:1/Fs:s.in_ch_size/Fs;
for i=1:s.in_ch_no
    input{i} = sin(2*pi*Fc*t+(i-1)*pi/2)*1024;
end
input{s.in_ch_no+1} = 2.4e9;
input{s.in_ch_no+2} = 30.72e6;
input{s.in_ch_no+3} = 18.0e6;
input{s.in_ch_no+4} = 'slow_attack';
input{s.in_ch_no+5} = 0;
input{s.in_ch_no+6} = 'slow_attack';
input{s.in_ch_no+7} = 0;
input{s.in_ch_no+8} = 2.4e9;
input{s.in_ch_no+9} = 30.72e6;
input{s.in_ch_no+10} = 18.0e6;

output = cell(1, s.out_ch_no + length(s.iio_dev_cfg.mon_ch));

for i = 1:5
output = stepImpl(s, input);
rssi1 = output{s.out_ch_no+1};
rssi2 = output{s.out_ch_no+2};
end

s.releaseImpl();

figure % new figure
ax1 = subplot(2,1,1); % top subplot
ax2 = subplot(2,1,2); % bottom subplot

plot(ax1,output{1});
title(ax1,'I');
xlabel('Sample');
ylabel('Amplitude');

plot(ax2,output{2});
title(ax2,'Q');
xlabel('Sample');
ylabel('Amplitude');

lclausen-adi, 2016-06-17:
Thanks, that looks OK. What does the configuration file you are using look like?

mustaphahaydar, 2016-06-17:
The configuration file ad9361.cfg 

data_in_device = cf-ad9361-dds-core-lpc
data_out_device = cf-ad9361-lpc
ctrl_device = ad9361-phy
channel = RX_LO_FREQ,IN,out_altvoltage0_RX_LO_frequency,
channel = RX_SAMPLING_FREQ,IN,in_voltage_sampling_frequency,
channel = RX_RF_BANDWIDTH,IN,in_voltage_rf_bandwidth,
channel = RX1_GAIN_MODE,IN,in_voltage0_gain_control_mode,
channel = RX1_GAIN,IN,in_voltage0_hardwaregain,
channel = RX1_RSSI,OUT,in_voltage0_rssi,
channel = RX2_GAIN_MODE,IN,in_voltage1_gain_control_mode,
channel = RX2_GAIN,IN,in_voltage1_hardwaregain,
channel = RX2_RSSI,OUT,in_voltage1_rssi,
channel = TX_LO_FREQ,IN,out_altvoltage1_TX_LO_frequency,
channel = TX_SAMPLING_FREQ,IN,out_voltage_sampling_frequency,
channel = TX_RF_BANDWIDTH,IN,out_voltage_rf_bandwidth,

lclausen-adi, 2016-06-17:
OK, that looks good as well.

What setup are you running on the ZED board? The ADI provided image, or something custom?

What is the output of iio_info when you run it on the device?

Thanks, Lars.

linkmauve, 2016-06-17:
I was going to open another bug report; this one is pretty much identical to my current issue, but if you prefer I could open another one. Here is a minimal testcase reproducing it in C: https://people.collabora.com/~emmanuel/libiio/testcase.c

My device is an mma8453 accelerometer, both iiod and the testcase are running on Linux. It fails with EPERM when ran as-is, or EINVAL if I comment out the channels retrieval/enabling (which is normal). This setup works fine when I’m using the `iio_channel_attr_read*()` family of functions, but then I can’t integrate it properly in an event loop.

I’ve been trying to debug why two connections are established in iiod, and also why the iio_device is corrupted when destroying the context, leading to a crash, but I’m not fully sure those issues are related, and I haven’t been able to determine any of that yet.

lclausen-adi, 2016-06-17:
That looks like a very different issue, please open a new report.

mustaphahaydar, 2016-06-17:
Hi,
sorry for the late reply. I am sure that the running setup on the zedboard  is  ADI provided image. And here is the output file (iio_info):
[iio_info.txt](https://github.com/analogdevicesinc/libiio/files/320963/iio_info.txt)

Thak you

mustaphahaydar, 2016-06-20:
Hi,
i used updating command (adi_update_tools.sh and adi_update_boot.sh), the iiod run automatically and the create_buffer its ok; Now i find the problem with push_buffer its blocking;
the buffer created is like empty, it possible the problem it with the speed of samples push because in linux is Hight-speed mode not enabled? or with the size of buffer? how can i solve it ?

Thank you

lclausen-adi, 2016-06-21:
Do you still get the warning about high-speed mode not being enabled?

lclausen-adi, 2016-06-21:
I just saw that your kernel built is from 2012. Can you try to upgrade to a newer version I think that should solve most of your issues. The easiest is to start with a fresh image. You can find the instructions for this here: https://wiki.analog.com/resources/tools-software/linux-software/embedded_arm_images

mustaphahaydar, 2016-06-21:
hi,
i tried to update version, he displayed ./adi_update_boot.sh is up to date. i will trying to do evrything from begining and i will see if its working or not.

Thank you

mustaphahaydar, 2016-06-21:
Hi 
Ca marcheeeeeeeeeeeeeee:)
its working now. it followed this steps 
Mount the FAT32 Boot partition by clicking on the 537MB Volume Icon on the desktop
Open a shell by clicking ion the Terminal Icon on the desktop (or CTRL+ALT+t)
Now convert the devictree.dtb into a dts file
Edit the file (mousepad or vi)
Convert the devictree.dts back into its binary format.
Unmount the file system (right click)

rgetz, 2016-06-21:
You mean :
https://wiki.analog.com/resources/eval/user-guides/ad-fmcomms2-ebz/software/linux/zynq_tips_tricks

-Robin

mustaphahaydar, 2016-06-21:
Yes

closed 2016-06-22 by pcercuei