analogdevicesinc/libiio #32: iio_create_buffer() failure in network mode

linkmauve, 2016-06-17:
(Reposting from #31)

Hello, I’m trying to use an mma8453 accelerometer in buffer mode, here is a minimal testcase to reproduce this issue: https://people.collabora.com/~emmanuel/libiio/testcase.c

Both iiod and the testcase are running on Linux, on two different devices. `iio_create_buffer()` fails with EPERM when the testcase is ran as-is, or with EINVAL if I comment out the channels retrieval/enabling (which is normal). This setup works fine when I’m using the `iio_channel_attr_read*()` family of functions, but then I can’t integrate it properly in an event loop. With the testcase ran as-is, `iiod` displays two different incoming connections; if I don’t create the buffer it only displays one.

Thanks for your help. :)

lclausen-adi, 2016-06-17:
Hi,

Have you checked that the user which runs the program has write access to the IIO devices? That would explain the EPERM.

Please also be aware that the network backend does not support non-blocking mode. If you want to run capture in-dependently from you main loop you'll have to use threads.

Lars

pcercuei, 2016-06-19:
If the program does not have write access to the IIO devices, the error would be EACCES, not EPERM, I think.

pcercuei, 2016-11-16:
Anything new on this? Otherwise I'll just close the bug.

closed 2017-01-11 by pcercuei