analogdevicesinc/libiio #33: Python bindings: memory leak due to circular references

jorenko, 2016-06-23:
The garbage collector is unable to determine what order to free a Context and its child objects in due to the circular reference between Context and Device, and Context's __del__ method. I've found that wrapping all parent-references in weakref.ref() solves this problem for me. The attached patch is against version 0.5 and shows my changes to make this work better.
[0000-weakref.patch.txt](https://github.com/analogdevicesinc/libiio/files/330596/0000-weakref.patch.txt)

pcercuei, 2016-07-05:
I will look into adding this patch. How can I verify that the circular dependencies are gone?

jorenko, 2016-07-05:
import gc
import iio
gc.set_debug(gc.DEBUG_LEAK)
c = iio.Context()
del c
gc.collect() # you should see messages about uncollectable objects here (or on the line above) if the leak still exists
print gc.garbage # this will be empty if the leak is fixed

pcercuei, 2016-07-05:
I'm trying to adapt it to the latest 0.7. I don't have messages about uncollectable objects, but "print gc.garbage" still shows data...

jorenko, 2016-07-05:
Whoops, I forgot that DEBUG_LEAK also turns on DEBUG_SAVEALL, which means /everything/ gets put into garbage. Try again without the set_debug(), and garbage will probably be empty then.

pcercuei, 2016-07-05:
I think I fixed it. Could you try again?

jorenko, 2016-07-06:
It works for me on the current master (https://github.com/analogdevicesinc/libiio/commit/9afd369ab2bf02cd26385567e27d5b80d76f730a)

pcercuei, 2016-07-06:
Ok, then I'll close the issue. Thanks!

closed 2016-07-06 by pcercuei