analogdevicesinc/libiio #35: Accessing already enabled device

xhpohanka, 2016-07-22:
When I open device that is already running (buffer/enable is 1) I cannot set its attributes with `iio_channel_attr_write()`.  This can happen eg. after when application closes unexpectedly.
For such case I'm lacking a possibility to disable buffer before calling `iio_device_create_buffer()`.

My issue can be workarounded by sequence `iio_device_create_buffer()`, `iio_buffer_destroy()`, `iio_device_create_buffer()` but it does not look very nice...

pcercuei, 2016-07-27:
That looks like a real bug. I'll look into it.

lclausen-adi, 2016-07-27:
It's a bit tricky to fix. Since we definitely dont want to disable all buffers when the context is created since this would upset any other running application using libiio. We could have a force_disable primitive that ensures that a specific buffer is disabled. But this should only work if the buffer is currently not opened by another application.

xhpohanka, 2016-07-27:
My current workaround lookslike following piece of code, 

    dev = iio_context_find_device(accel.ctx, ACCEL_NAME);
    if (dev == NULL) {
        iio_context_destroy(accel.ctx);
        return -1;
    }

	iio_device_attr_write_longlong(dev, "buffer/enable", 0);

I thing that it is acceptable however at first I searched for something like `iio_device_buffer_disable()`.

qo0op, 2020-05-11:
Currently I'm facing the same/similar issue as @xhpohanka. When application is closed unexpectedly, buffer stays enabled. (When I rerun app and trying to refill buffer using "iio_buffer_refill" it returns err -11).

rgetz, 2020-06-04:
You mean when the application crashes? We added a stress test, but it closes things properly, but I can add something that randomly kills threads to see if I can reproduce things.