analogdevicesinc/libiio #47: Libiio release for Windows

damercer, 2016-10-24:
Is the Windows installer release for 32 or 64 bits?

If it is for 64 bits, it would be helpful if the file name indicated that.
Also if it is for 64 bits, would it be possible for you to provide a 32 bit version as well?
( Use will be will be with 32 bit Python ).

Thanks

pcercuei, 2016-10-24:
The Windows installer is for both 32-bit and 64-bit versions.
On 64-bit systems, it will install the 32-bit version too.

damercer, 2016-10-24:
So there are two different .dll files? What are their names and where are they installed?
Because the one I found seems to be for 64 bits. 32 bit Python says it is not a WIN32 file.

Thanks

pcercuei, 2016-10-25:
C:\Windows\System32\libiio.dll is the 64-bit version
C:\Windows\SysWOW64\libiio.dll is the 32-bit version

Can I close this bug?

damercer, 2016-10-25:
Python 32 bit seems to find the "right" version of libiio.dll but it fails to properly load see issue #48.
However I don't see a file named libiio.dll in C:\Windows\SysWOW64\?

closed 2016-10-25 by damercer

pcercuei, 2016-10-25:
It might have been added after the version 2015.02. Try with a nighty build of libiio:
https://ci.appveyor.com/project/analogdevicesinc/libiio/build/artifacts?branch=master