analogdevicesinc/libiio #48: Using Libiio with Python

damercer, 2016-10-24:
The Windows installer places the libiio.dll file in the system32 directory but Python (iio.py) does not seem
to find it there. Installer probably needs to add/change something in a Path?

Thanks

damercer, 2016-10-24:
Python ( 2.7 ) can't load libiio.dll library file ( 32 bit version from release 2015.2 Feb 22 ). See trace back message:

>>> import ctypes
>>> ctypes.cdll.LoadLibrary('libiio.dll')

Traceback (most recent call last):
  File "<pyshell#1>", line 1, in <module>
    ctypes.cdll.LoadLibrary('libiio.dll')
  File "C:\Python27\lib\ctypes\__init__.py", line 443, in LoadLibrary
    return self._dlltype(name)
  File "C:\Python27\lib\ctypes\__init__.py", line 365, in __init__
    self._handle = _dlopen(self._name, mode)
WindowsError: [Error 127] The specified procedure could not be found
>>> 

Thanks

pcercuei, 2016-10-25:
So I just tested it here, with Python 2.7.11 64-bit, and it works fine.

The error "The specified procedure could not be found" might be because your version of libiio is too old, while you're using iio.py from git master (I assume). Could you try with a build of libiio from appveyor?

Thanks,
- Paul

damercer, 2016-10-25:
I downloaded and  installed from this on appveyor: 2015.1.516

Ran iio_info.py and got this back which I think is the same error:

Python 2.7.11 (v2.7.11:6d1b6a68f775, Dec  5 2015, 20:32:19) [MSC v.1500 32 bit (Intel)] on win32
Type "copyright", "credits" or "license()" for more information.
>>> 
============== RESTART: C:\My_Documents\ALM\iio-lib\iio_info.py ==============

Traceback (most recent call last):
  File "C:\My_Documents\ALM\iio-lib\iio_info.py", line 16, in <module>
    import iio
  File "C:\My_Documents\ALM\iio-lib\iio.py", line 49, in <module>
    _lib = _cdll.LoadLibrary('libiio.dll' if 'Windows' in _system() else 'libiio.so.0')
  File "C:\Python27\lib\ctypes\__init__.py", line 443, in LoadLibrary
    return self._dlltype(name)
  File "C:\Python27\lib\ctypes\__init__.py", line 365, in __init__
    self._handle = _dlopen(self._name, mode)
WindowsError: [Error 126] The specified module could not be found
>>>

pcercuei, 2016-10-26:
Could you try with the 64-bit version of Python?
Just to bisect the issue.

damercer, 2016-10-26:
I could I suppose. I only have access to one 64 bit machine ( the ADI laptop ). The ones I do most of my work on are 32 bit. I've been avoiding taking the time to switch versions on the one machine.

Thanks

pcercuei, 2016-10-26:
Oh, I thought you were using Python 32-bits on a 64-bits machine.
On 32-bit systems, C:\Windows\System32\libiio.dll is the 32-bit version, not the 64-bit version.

The error you report now is different: it's Error 126, it was Error 127 before.
What does _system() actually return?

-Paul

damercer, 2016-10-26:
The first error report was for 32 bit Python on a 32 bit processor.
The second report was for 32 bit Python on the 64 bit ADI laptop.
Identical Python installations in both cases.
Same line in ctypes_init.py", line 365, in both cases.
The different numbers might be because in the first case I simply typed
the ctypes command to load the file in the Python shell where I ran the
program in the second case?

pcercuei, 2016-10-27:
I just tried with the 32-bit version of Python. It just works...
If you run "iio_info.exe" in a console what does it give you as the library version?

damercer, 2016-10-27:
I typed iio_info.exe in a Windows command prompt window and it said the program can't run because MSVCR120.dll is missing. This missing file might be the root cause? At any rate something in the installer should have checked for this required file and installed it too? ( or at least informed the user where and how to do it ).

Thanks

rgetz, 2016-10-27:
Doug:

You are right - it should be included in the installer. 

It is part of the Visual C++ Redistributable Packages for Visual Studio. For now - can you install that?
https://www.microsoft.com/en-us/download/details.aspx?id=40784

damercer, 2016-10-27:
I just downloaded and installed Visual C++ Redistributable Packages for Visual Studio 2013 on the ADI lap-top. ( 64 bit version , 32 bit version installed first but it didn't work )

Typing iio_info.exe prints the following:

C:\My_Documents\ALM\iio-lib>iio_info.exe
Library version: 0.7 (git tag: 7d673a2)
Unable to create IIO context

Running the Python ii0_info.py program still results in the same error as before.

Thanks

damercer, 2016-10-31:
Here is some additional information about this issue. I have access to three different Windows computers and the following is what happens on the third.
System Info:
Windows 8.1
Intel Atom CPU ( Quad Core ) Z3735F
32-bit Operating System, x64 based processor

I got the same missing file error the first time I ran iio_info.exe as expected.
I down loaded and installed the 2013 redistribution package. 
Only the 32-bit x86 version would load because the OS is 32-bits.

Now iio_info.exe returns error, file MSVCR120D.dll missing. Note the D in the file name.

Thanks 

Doug

rgetz, 2016-10-31:
If it’s missing MSVCR*D.dll, it’s a debug build.  The debug runtime is not part of the redistributable package -- just the release versions. 

That is a different problem in our build (thanks for reporting). 

@pcercuei / @radhermit : are we building with debug? or release? any ideas?

Thanks
-Robin

pcercuei, 2016-10-31:
We build in release mode: https://github.com/analogdevicesinc/libiio/blob/master/appveyor.yml#L28

damercer, 2016-10-31:
Interesting but why doesn't it report that that file is missing on the other two computes? With the ( more or less ) same redistribution package.

radhermit, 2016-10-31:
We're only allowed to redistribute the non-debug versions of the MSVC dlls which I'd suggest doing if they're required so people don't have to hunt down the related installers themselves.

Also, the appveyor setup is currently building both 32 and 64 bit versions and the installer looks like it's set to install the version related to the system it's running on.

damercer, 2016-11-01:
Why does iio_info.exe work on the Windows 7 ADI laptop but not on my Windows 8.1 ( 32bit OS ) netbook? AppVeyor builds 514, 516 and 520 from master all fail saying that MSVCR120D.dll is missing. On the ADI laptop it works without apparently needing that file.

The installer is not installing 32 bit versions of iio_info.exe ( or the other .exe files ) on 32 bit systems. ( Even if you think it is supposed to ).

Also, all this checking has not solved my Python issue.......

Thanks

Doug

pcercuei, 2017-03-31:
I fixed the dependency on MSVCR120D.dll in the 32-bit version of libiio; could you try again?

damercer, 2017-03-31:
Yes, It would seem to be doing something now. With the M2K plugged in 
and running the iio_info.py program returns a long list with 15 devices 
found. I'm guessing this would be correct for the M2K.

Doug

pcercuei, 2017-04-01:
Sounds like it works, yes. I'll close this bug then.

closed 2017-04-01 by pcercuei