analogdevicesinc/libiio #49: libiio crash with osc

rgetz, 2016-10-26:
When using ./osc with Pluto, sometimes I get this:

Thread 1 "osc" received signal SIGSEGV, Segmentation fault.
0x00007ffff4cbb3a6 in iio_channel_attr_write_raw ()
   from /usr/lib/x86_64-linux-gnu/libiio.so.0
(gdb) bt
#0  0x00007ffff4cbb3a6 in iio_channel_attr_write_raw ()
   from /usr/lib/x86_64-linux-gnu/libiio.so.0
#1  0x00007ffff4cbbd66 in iio_channel_attr_write_double ()
   from /usr/lib/x86_64-linux-gnu/libiio.so.0
#2  0x00007ffff7bbed47 in iio_widget_save (widget=0x7fffd008e378)
    at iio_widget.c:357
#3  0x00007ffff7bbedd8 in spin_button_progress_step (iio_w=0x7fffd008e378)
    at iio_widget.c:518
#4  0x00007ffff5625fc3 in ?? () from /lib/x86_64-linux-gnu/libglib-2.0.so.0
#5  0x00007ffff562554a in g_main_context_dispatch ()
   from /lib/x86_64-linux-gnu/libglib-2.0.so.0
#6  0x00007ffff5625900 in ?? () from /lib/x86_64-linux-gnu/libglib-2.0.so.0
#7  0x00007ffff5625c22 in g_main_loop_run ()
   from /lib/x86_64-linux-gnu/libglib-2.0.so.0
#8  0x00007ffff725e567 in gtk_main ()
   from /usr/lib/x86_64-linux-gnu/libgtk-x11-2.0.so.0
#9  0x0000000000402e4f in main (argc=<optimized out>, argv=<optimized out>)
    at oscmain.c:237

Where should the pointer checking be done? in libiio or osc?
I think this is done with the version of libiio in the deb repository. (libiio-dev), not trunk.

-Robin

lclausen-adi, 2016-10-26:
Can you install libiio with debug symbols (libiio0-dbgsym package)? This will improve the quality of the backtrace.

rgetz, 2016-10-26:
analog@imhotep:~$ sudo apt-get install libiio0-dbgsym
Reading package lists... Done
Building dependency tree       
Reading state information... Done
E: Unable to locate package libiio0-dbgsym

Oh - I need to add debian-debug to my /etc/apt/sources.list - doing that, and apt-get update, then it works (only added that for those who run into the same issue).

After that, I get:

Thread 1 "osc" received signal SIGSEGV, Segmentation fault.
0x00007ffff7bbfd77 in spin_button_progress_step (iio_w=0x7fffc8043d88) at iio_widget.c:509
509		if (pdata->progress < 1.0) {
(gdb) bt
#0  0x00007ffff7bbfd77 in spin_button_progress_step (iio_w=0x7fffc8043d88) at iio_widget.c:509
#1  0x00007ffff5626103 in ?? () from /lib/x86_64-linux-gnu/libglib-2.0.so.0
#2  0x00007ffff562568a in g_main_context_dispatch () from /lib/x86_64-linux-gnu/libglib-2.0.so.0
#3  0x00007ffff5625a40 in ?? () from /lib/x86_64-linux-gnu/libglib-2.0.so.0
#4  0x00007ffff5625d62 in g_main_loop_run () from /lib/x86_64-linux-gnu/libglib-2.0.so.0
#5  0x00007ffff7260447 in gtk_main () from /usr/lib/x86_64-linux-gnu/libgtk-x11-2.0.so.0
#6  0x0000000000402e4f in main (argc=<optimized out>, argv=<optimized out>) at oscmain.c:237

dNechita, 2016-10-26:
Also can you please recompile osc like this:
- open Makefile, locate "-O2" and replace it with "-ggdb3"
- make clean && make -j2
This will produce a more detailed backtrace.

rgetz, 2016-10-26:
-ggdb3 doesn't make a difference when -g is already there. I replaced -O2 with -O0.

Thread 1 "osc" received signal SIGSEGV, Segmentation fault.
0x00007ffff4cac3aa in iio_channel_attr_write_raw (chn=0x7fffd005efe0, attr=0x7ffff7bcff84 "phase", src=0x7fffffffdad0, len=13)
    at /build/libiio-pJcb2D/libiio-0.7/channel.c:358
358	/build/libiio-pJcb2D/libiio-0.7/channel.c: No such file or directory.
(gdb) bt
#0  0x00007ffff4cac3aa in iio_channel_attr_write_raw (chn=0x7fffd005efe0, attr=0x7ffff7bcff84 "phase", src=0x7fffffffdad0, len=13)
    at /build/libiio-pJcb2D/libiio-0.7/channel.c:358
#1  0x00007ffff4cac3eb in iio_channel_attr_write (chn=<optimized out>, attr=<optimized out>, src=<optimized out>)
    at /build/libiio-pJcb2D/libiio-0.7/channel.c:368
#2  0x00007ffff4cacd66 in iio_channel_attr_write_double (chn=0x7fffd005efe0, attr=0x7ffff7bcff84 "phase", val=<optimized out>)
    at /build/libiio-pJcb2D/libiio-0.7/channel.c:691
#3  0x00007ffff7bba98a in spin_button_save (widget=0x7fffd008df78, is_double=true) at iio_widget.c:147
#4  0x00007ffff7bbaa26 in iio_spin_button_savedbl (widget=0x7fffd008df78) at iio_widget.c:164
#5  0x00007ffff7bbb284 in iio_widget_save (widget=0x7fffd008df78) at iio_widget.c:357
#6  0x00007ffff7bbb89e in spin_button_progress_step (iio_w=0x7fffd008df78) at iio_widget.c:518
#7  0x00007ffff5618103 in ?? () from /lib/x86_64-linux-gnu/libglib-2.0.so.0
#8  0x00007ffff561768a in g_main_context_dispatch () from /lib/x86_64-linux-gnu/libglib-2.0.so.0
#9  0x00007ffff5617a40 in ?? () from /lib/x86_64-linux-gnu/libglib-2.0.so.0
#10 0x00007ffff5617d62 in g_main_loop_run () from /lib/x86_64-linux-gnu/libglib-2.0.so.0
#11 0x00007ffff7252447 in gtk_main () from /usr/lib/x86_64-linux-gnu/libgtk-x11-2.0.so.0
#12 0x0000000000403921 in main (argc=3, argv=0x7fffffffe288) at oscmain.c:237

pcercuei, 2016-10-27:
Do you know what "double" value it tries to print? (in iio_channel_attr_write_double)
It doesn't appear in the backtrace.

rgetz, 2016-10-27:
Here is a different trace, same setup.

Thread 1 "osc" received signal SIGSEGV, Segmentation fault.
0x00007ffff4cac386 in iio_channel_attr_read (chn=0x7fffdc02f100, 
    attr=0x7ffff7bcff7e "scale", dst=0x7fffffffdb80 "\260!e", len=1024)
    at /build/libiio-pJcb2D/libiio-0.7/channel.c:348
348	/build/libiio-pJcb2D/libiio-0.7/channel.c: No such file or directory.
(gdb) bt
#0  0x00007ffff4cac386 in iio_channel_attr_read (chn=0x7fffdc02f100, 
    attr=0x7ffff7bcff7e "scale", dst=0x7fffffffdb80 "\260!e", len=1024)
    at /build/libiio-pJcb2D/libiio-0.7/channel.c:348
#1  0x00007ffff4cacc5c in iio_channel_attr_read_double (chn=<optimized out>, 
    attr=<optimized out>, val=0x7fffffffdfd0)
    at /build/libiio-pJcb2D/libiio-0.7/channel.c:666
#2  0x00007ffff7bc7c13 in save_scale_widget_value (data=0x7fffdc021db8)
    at plugins/dac_data_manager.c:1323
#3  0x00007ffff7bbb8bc in spin_button_progress_step (iio_w=0x7fffdc021e30)
    at iio_widget.c:520
#4  0x00007ffff5618103 in ?? () from /lib/x86_64-linux-gnu/libglib-2.0.so.0
#5  0x00007ffff561768a in g_main_context_dispatch ()
   from /lib/x86_64-linux-gnu/libglib-2.0.so.0
#6  0x00007ffff5617a40 in ?? () from /lib/x86_64-linux-gnu/libglib-2.0.so.0
#7  0x00007ffff5617d62 in g_main_loop_run ()
   from /lib/x86_64-linux-gnu/libglib-2.0.so.0
#8  0x00007ffff7252447 in gtk_main ()
   from /usr/lib/x86_64-linux-gnu/libgtk-x11-2.0.so.0
#9  0x0000000000403921 in main (argc=3, argv=0x7fffffffe308) at oscmain.c:237

(gdb) print *chn
$1 = {dev = 0x7fffdc02e560, pdata = 0x0, 
  userdata = 0x7ffff58df960 <g_idle_funcs>, is_output = false, 
  is_scan_element = false, format = {length = 6751152, bits = 0, shift = 0, 
    is_signed = false, is_fully_defined = false, is_be = false, 
    with_scale = false, scale = 4.1946173331921832e-321}, name = 0x0, 
  id = 0x0, index = 0, modifier = IIO_NO_MOD, type = IIO_VOLTAGE, attrs = 0x0, 
  nb_attrs = 0}
(gdb) print *chn->dev
$2 = {ctx = 0x140, pdata = 0x74, userdata = 0x7fffdc02f480, name = 0x0, 
  id = 0x7ffff58df960 <g_idle_funcs> "\200;a\365\377\177", attrs = 0x0, 
  nb_attrs = 6751152, debug_attrs = 0x0, nb_debug_attrs = 697, channels = 0x0, 
  nb_channels = 0, mask = 0x0, words = 0}
(gdb) print *chn->dev->name
Cannot access memory at address 0x0

rgetz, 2016-10-27:
And sometimes here:

Thread 1 "osc" received signal SIGSEGV, Segmentation fault.
0x00007ffff7bbb7b8 in spin_button_progress_step (iio_w=0x7fffd00015d8) at iio_widget.c:507
507		void (*on_complete_cb)(void *) = pdata->on_complete;
(gdb) bt
#0  0x00007ffff7bbb7b8 in spin_button_progress_step (iio_w=0x7fffd00015d8) at iio_widget.c:507
#1  0x00007ffff5618103 in ?? () from /lib/x86_64-linux-gnu/libglib-2.0.so.0
#2  0x00007ffff561768a in g_main_context_dispatch () from /lib/x86_64-linux-gnu/libglib-2.0.so.0
#3  0x00007ffff5617a40 in ?? () from /lib/x86_64-linux-gnu/libglib-2.0.so.0
#4  0x00007ffff5617d62 in g_main_loop_run () from /lib/x86_64-linux-gnu/libglib-2.0.so.0
#5  0x00007ffff7252447 in gtk_main () from /usr/lib/x86_64-linux-gnu/libgtk-x11-2.0.so.0
#6  0x0000000000403921 in main (argc=3, argv=0x7fffffffe2f8) at oscmain.c:237
(gdb) print *pdata
Cannot access memory at address 0x0

pcercuei, 2016-10-27:
Well it looks like that the OSC is passing invalid iio_channel pointers to libiio...

rgetz, 2016-10-27:
or is it libiio giving invalid channels to osc to process? :)

Something is getting clobbered somewhere. The weird thing is - it only crashes if I don't load a profile in osc (osc -p -). If I let a profile load, it works without issue.

-Robin

rgetz, 2016-10-27:
I also can try trunk libiio, and see if that changes anything (if that helps?)

-Robin

pcercuei, 2016-10-28:
It's Never Libiio™

Does it only happen with Pluto? What do you do to trigger the bug?

radhermit, 2016-11-03:
This might be fixed via the latest master in osc now after I pushed a few network context fixes, at least I can't reproduce it in the same manner anymore.

rgetz, 2016-11-04:
Yes - I can't reproduce it now. 

I will try some more, and open something in the osc project if it happens again. 

- Robin

closed 2016-11-04 by rgetz