analogdevicesinc/libiio #50: Installing libiio breaks pysmu

damercer, 2016-11-05:
Having libiio installed causes pysmu to break.

Python Error when libiio is installed:

Python 2.7.12 (v2.7.12:d33e0cf91556, Jun 27 2016, 15:19:22) [MSC v.1500 32 bit (Intel)] on win32
Type "copyright", "credits" or "license()" for more information.
============= RESTART: D:\ALM\ALM Software\alice-desktop-1.1.pyw =============

Traceback (most recent call last):
  File "D:\ALM\ALM Software\alice-desktop-1.1.pyw", line 8, in <module>
    from pysmu import *
  File "C:\Python27\lib\site-packages\pysmu\__init__.py", line 13, in <module>
    import _pysmu
ImportError: DLL load failed: The specified module could not be found.

This happened on all three of my Windows computers. Uninstalling libiio fixes the problem and pysmu will import properly again.

Just thought you would like to know....

Doug

radhermit, 2016-11-15:
It looks like libsmu finds (and uses) libiio's libusb-1.0.dll first if both are installed and doesn't work properly due to that. Just to note, libsmu still uses a custom libusb build for windows hotplugging support which is hopefully being merged upstream in the next release (1.0.22).

Our options are probably the following:

1. figure out how to make libsmu use the correct search order for finding its own dlls
2. moving libiio dlls out of the main system folders that get fallen back on when searching
3. revert to linking libusb in statically for libsmu

pcercuei, 2016-11-16:
Libiio's installer doesn't overwrite libusb if it's already installed: https://github.com/analogdevicesinc/libiio/blob/master/libiio.iss.cmakein#L59
So that's strange.

rgetz, 2016-11-16:
Paul:

libsmu still uses a custom libusb build, and puts it in the same folder as the application. So when libiio looks for it - it doesn't see it, -and it installs it in the system folder, which then libsmu finds and uses. 

Does that make more sense?

pcercuei, 2016-11-16:
Yes and no; the DLLs that are in the same folder as the application take precedence. Maybe it's different for Python though.

radhermit, 2016-11-16:
Right, currently python doesn't know to look in the application path at all so option 4 would be to try modifying the syspath in the libsmu python bindings before the shared module that links with libusb is loaded so it would load the right libusb library.

pcercuei, 2016-11-16:
Yet another option would be to link libusb statically to libiio, if that works I'm fine with it.

radhermit, 2016-11-16:
That should work as well unless the libusb dll was installed in the system path by something else. I'll try to get python to search its own application path first until the windows hotplugging stuff is merged and released upstream.

radhermit, 2016-11-17:
I went with linking libusb in statically for libsmu. The latest builds for master and related branches should work now installed alongside libiio. If you have issues open a new bug for libsmu.

closed 2016-11-17 by radhermit