analogdevicesinc/libiio #51: cmake on OS-X spews warnings

rgetz, 2016-11-07:
adi-mm:build analogdevices$ uname -a
Darwin adi-mm.local 15.6.0 Darwin Kernel Version 15.6.0: Thu Sep  1 15:01:16 PDT 2016; root:xnu-3248.60.11~2/RELEASE_X86_64 x86_64

adi-mm:github analogdevices$ cd libiio/
adi-mm:libiio analogdevices$ mkdir build
adi-mm:libiio analogdevices$ cd build/
adi-mm:build analogdevices$ cmake ../

CMake Warning (dev):
  Policy CMP0042 is not set: MACOSX_RPATH is enabled by default.  Run "cmake
  --help-policy CMP0042" for policy details.  Use the cmake_policy command to
  set the policy and suppress this warning.

  MACOSX_RPATH is not specified for the following targets:

   iio

This warning is for project developers.  Use -Wno-dev to suppress it.

-- Generating done
-- Build files have been written to: /Users/analogdevices/github/libiio/build

adi-mm:build analogdevices$ cmake --help-policy CMP0042
CMP0042
-------

``MACOSX_RPATH`` is enabled by default.

CMake 2.8.12 and newer has support for using ``@rpath`` in a target's install
name.  This was enabled by setting the target property
``MACOSX_RPATH``.  The ``@rpath`` in an install name is a more
flexible and powerful mechanism than ``@executable_path`` or ``@loader_path``
for locating shared libraries.

CMake 3.0 and later prefer this property to be ON by default.  Projects
wanting ``@rpath`` in a target's install name may remove any setting of
the ``INSTALL_NAME_DIR`` and ``CMAKE_INSTALL_NAME_DIR``
variables.

This policy was introduced in CMake version 3.0.  CMake version
3.5.0 warns when the policy is not set and uses OLD behavior.  Use
the cmake_policy command to set it to OLD or NEW explicitly.

.. note::
  The ``OLD`` behavior of a policy is
  ``deprecated by definition``
  and may be removed in a future version of CMake.

rgetz, 2016-11-07:
And then 'make' doesn't work:
adi-mm:build analogdevices$ make
Scanning dependencies of target iio
[  4%] Building C object CMakeFiles/iio.dir/channel.c.o
[  8%] Building C object CMakeFiles/iio.dir/device.c.o
[ 12%] Building C object CMakeFiles/iio.dir/context.c.o
[ 16%] Building C object CMakeFiles/iio.dir/buffer.c.o
[ 20%] Building C object CMakeFiles/iio.dir/utilities.c.o
[ 25%] Building C object CMakeFiles/iio.dir/scan.c.o
[ 29%] Building C object CMakeFiles/iio.dir/usb.c.o
[ 33%] Building C object CMakeFiles/iio.dir/network.c.o
[ 37%] Building C object CMakeFiles/iio.dir/xml.c.o
[ 41%] Building C object CMakeFiles/iio.dir/lock.c.o
[ 45%] Building C object CMakeFiles/iio.dir/iiod-client.c.o
[ 50%] Linking C shared library iio.framework/iio
Copying OS X content iio.framework/Versions/0.8/Headers/iio.h
[ 50%] Built target iio
Scanning dependencies of target libiio-pkg
[ 54%] Generating libiio-0.8-temp.pkg
pkgbuild: error: Specified root path "/Users/analogdevices/github/libiio/build/Debug" does not exist.
make[2]: *** [libiio-0.8-temp.pkg] Error 1
make[1]: *** [CMakeFiles/libiio-pkg.dir/all] Error 2
make: *** [all] Error 2

rgetz, 2016-11-07:
making the dir, gets past that - but then still errors out:
adi-mm:build analogdevices$ mkdir Debug
adi-mm:build analogdevices$ make
[ 50%] Built target iio
[ 54%] Generating libiio-0.8-temp.pkg
pkgbuild: Inferring bundle components from contents of /Users/analogdevices/github/libiio/build/Debug
pkgbuild: Wrote package to /Users/analogdevices/github/libiio/build/libiio-0.8-temp.pkg
[ 58%] Generating libiio-0.8.pkg
productbuild: Wrote product to /Users/analogdevices/github/libiio/build/libiio-0.8.pkg
[ 58%] Built target libiio-pkg
Scanning dependencies of target iio_reg
[ 62%] Building C object tests/CMakeFiles/iio_reg.dir/iio_reg.c.o
[ 66%] Linking C executable iio_reg
[ 66%] Built target iio_reg
Scanning dependencies of target iio_readdev
[ 70%] Building C object tests/CMakeFiles/iio_readdev.dir/iio_readdev.c.o
/Users/analogdevices/github/libiio/tests/iio_readdev.c:147:9: warning: implicit
      declaration of function 'sigwaitinfo' is invalid in C99
      [-Wimplicit-function-declaration]
                ret = sigwaitinfo(mask, NULL);
                      ^
1 warning generated.
[ 75%] Linking C executable iio_readdev
Undefined symbols for architecture x86_64:
  "_sigwaitinfo", referenced from:
      _sig_handler_thd in iio_readdev.c.o
ld: symbol(s) not found for architecture x86_64
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make[2]: *** [tests/iio_readdev] Error 1
make[1]: *** [tests/CMakeFiles/iio_readdev.dir/all] Error 2
make: *** [all] Error 2

rgetz, 2016-11-07:
Adding this in:

https://www.snip2code.com/Snippet/794088/sigwaitinfo()-stub-for-OS-X

seemed to help (or at least makes iio_info and iio_readdev work.

pcercuei, 2016-11-07:
For the first issue it looks like it's because of this: https://github.com/analogdevicesinc/libiio/blob/master/CMakeLists.txt#L321
Try to remove the /Debug suffix.

I'll integrate the stub.

rgetz, 2016-11-07:
Yes - that fixes the /Users/analogdevices/github/libiio/build/Debug error, still has the MACOSX_RPATH is not specified issue.

and I can confirm it does build:
-rw-r--r--   1 analogdevices  staff  213723 Nov  7 17:07 libiio-0.8-temp.pkg
-rw-r--r--   1 analogdevices  staff  222240 Nov  7 17:07 libiio-0.8.pkg

When I install it from the GUI (by double clicking on it), I get a warning:
This package is incompatible with this version of OS X and may fail to install.
Installing this package may damage your system, and the installation may fail. To stop the program and the installation, click “Cancel Installation”.

And eventually a "Installation Failed".

rgetz, 2016-11-08:
https://travis-ci.org/analogdevicesinc/libiio

Which currently fails on osx.

pcercuei, 2016-11-16:
Fixed in latest master.

closed 2016-11-16 by pcercuei