analogdevicesinc/libiio #52: Make install errors on OS X

rgetz, 2016-11-07:
adi-mm:build analogdevices$ sudo make install
Password:
[ 50%] Built target iio
[ 58%] Built target libiio-pkg
[ 66%] Built target iio_reg
[ 75%] Built target iio_readdev
[ 83%] Built target iio_info
[ 91%] Built target iio_adi_xflow_check
[100%] Built target iio_genxml
Install the project...
-- Install configuration: "RelWithDebInfo"
-- Installing: /usr/lib/pkgconfig/libiio.pc
CMake Error at cmake_install.cmake:31 (file):
  file INSTALL cannot copy file
  "/Users/analogdevices/github/libiio/build/libiio.pc" to
  "/usr/lib/pkgconfig/libiio.pc".

make: *** [install] Error 1

my understanding is that this is part of System Integrity Protection (SIP, sometimes referred to as rootless) is a security feature of OS X.

I think they now go into /usr/local (which isn't protected.)
https://support.apple.com/en-us/HT204899

pcercuei, 2016-11-07:
"make libiio-pkg"

rgetz, 2016-11-07:
It could have  -- I didn't really look -- Just learning OS-X as I go... :(

so:
adi-mm:build analogdevices$ make libiio-pkg

and then:

adi-mm:build analogdevices$ sudo /usr/sbin/installer -pkg ./libiio-0.8.pkg -target /
installer: Package name is Libiio
installer: Upgrading at base path /
installer: The upgrade was successful.
adi-mm:build analogdevices$ 

seemed to work.

rgetz, 2016-11-07:
If that is the preferred method for OS-X, I will update the docs, and close this.

rgetz, 2016-11-07:
OK - that's not it...

adi-mm:build analogdevices$ cmake ../
-- The C compiler identification is AppleClang 8.0.0.8000042
-- Check for working C compiler: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/cc
-- Check for working C compiler: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
CMake Error: The following variables are used in this project, but they are set to NOTFOUND.
Please set them or make sure they are set and tested correctly in the CMake files:
LIBIIO_INCLUDEDIR
   used as include directory in directory /Users/analogdevices/github/libad9361-iio
   used as include directory in directory /Users/analogdevices/github/libad9361-iio
   used as include directory in directory /Users/analogdevices/github/libad9361-iio
   used as include directory in directory /Users/analogdevices/github/libad9361-iio
   used as include directory in directory /Users/analogdevices/github/libad9361-iio
   used as include directory in directory /Users/analogdevices/github/libad9361-iio
   used as include directory in directory /Users/analogdevices/github/libad9361-iio
LIBIIO_LIBRARIES
    linked by target "ad9361" in directory /Users/analogdevices/github/libad9361-iio

-- Configuring incomplete, errors occurred!
See also "/Users/analogdevices/github/libad9361-iio/build/CMakeFiles/CMakeOutput.log".

pcercuei, 2016-11-16:
Fixed in latest master. Clang has some trouble detecting frameworks in OSX Mavericks, but everything works smooth with El Capitán.

closed 2016-11-16 by pcercuei