analogdevicesinc/libiio #54: Include MATLAB Bindings in installer

mfornero, 2016-11-08:
It appears that the MATLAB bindings (and critically the header iio-wrapper.h) are not included with the installers / zip files

https://ci.appveyor.com/project/analogdevicesinc/libiio/build/2015.1.528/artifacts

pcercuei, 2016-11-08:
They are not built either. We'd need Matlab installed on AppVeyor, I don't know how easy it would be.

mfornero, 2016-11-08:
I think as an initial step, simply including the bindings/matlab directory in the generated .zip files would be useful

mfornero, 2016-11-08:
Actually "building" on AppVeyor likely wouldn't help-- I believe the current CMake scripts run on the build machine, which wouldn't tell us anything about where MATLAB is on the install machine.

rgetz, 2016-11-08:
So, these rules needs to be in the installer, not in the make system?

mfornero, 2016-11-08:
Right-- if you're planning to distribute a binary installer, it would need to know how to find / run MATLAB.

Alternatively, it could be packaged as a MATLAB app (see https://www.mathworks.com/help/matlab/creating_guis/app-creation.html)

rgetz, 2016-11-08:
Does MathWorks offer a CI SaaS? Like Travis or Appveour? For MATLAB apps.

mfornero, 2016-11-08:
Looks like some people have integrated MATLAB into various CI services, but I'm not familiar with a specific offering like Appveyor

http://blogs.mathworks.com/developer/category/continuous-integration/

radhermit, 2016-11-11:
I'm pretty sure we'll have to integrate MATLAB bindings testing with our own internal build server for now since I don't think a service exists like that yet.

In terms of windows integration, the remaining piece under the current installer method would be to write/find/modify some pascal-based module for innosetup to run the equivalent of what the cmake script for the bindings does to install the files on Linux.

rgetz, 2016-11-11:
I think we determined that attempting to do this at build time - doesn't really work - since end users will have the MATLAB install location at a different place. 

Matt: is their an installer solution that you have?

radhermit, 2016-11-11:
I think you were misunderstanding my comment. The pascal script would get executed by the installer under the current install setup when it is run by the user at install time. This would be similar to how the modpath module is used by libsmu to add/remove the application path to the system path.

radhermit, 2016-11-11:
Simply bundling the matlab bindings in the zip file artifacts will be done when #61 is merged.

rgetz, 2016-12-17:
Tim:

>The pascal script would get executed by the 
>installer under the current install setup when it 
>is run by the user at install time.

Understand- the question for mfornero, does he know of an project that does that so we don't need to create the pascal ourselves ;) (just trying to save some time/not do it wrong).

- Robin

rgetz, 2016-12-20:
Closing this, since the pull request #61 was merged.

closed 2016-12-20 by rgetz