analogdevicesinc/libiio #57: libiio needs to install some udev rules

rgetz, 2016-11-10:
analog@imhotep:~/github/libiio/build$ iio_info -s
Library version: 0.8 (git tag: 6a2c22a)
No contexts found.

analog@imhotep:~/github/libiio/build$ sudo iio_info -s
Library version: 0.8 (git tag: 6a2c22a)
Available contexts:
	0: 0456:b673 (Analog Devices Inc. PlutoSDR (ADALM-PLUTO)) [usb:3.8.5]

USB IIO devices should be open read/write to normal users.

pcercuei, 2016-11-10:
I agree on the principle, but the rule should not be based on the VID/PID of the USB devices, because then it would only work with ADI products. The detection should be done on the presence of the "IIO" string in the interface descriptors; but I don't know if that's possible/easy to do.

lclausen-adi, 2016-11-10:
ATTRS{interface}=="IIO"

But maybe we want to put additional restrictions on it as well (like the PID/VID) so that the rule is not too broad.

rgetz, 2016-11-10:
@pcercuei : Everyone should be able to fork, and set pull requests for their products - So, I'm less concerned about other vendors, we will gladly accept patches -- it does complicate the release process - releasing hardware becomes dependent on updating the library, even if the library doesn't need to be updated - so I understand that.

What would be the best thing? To make each vendor specify their own .spec or .deb with libiio as a prerequisite? So Pluto would have it's own, which would install it's own pid/vid udev rule?

?

rgetz, 2016-11-10:
Maybe start a new repo, with just udev rules for Analog Devices hardware?

mhennerich, 2016-11-10:
The udev rule for PlutoSDR exists here: 
https://github.com/analogdevicesinc/plutosdr-fw/blob/master/scripts/53-adi-plutosdr-usb.rules

-Michael

rgetz, 2016-11-10:
So, doing:

analog@imhotep:/$ cd /etc/udev/rules.d/
analog@imhotep:/etc/udev/rules.d$ sudo wget https://raw.githubusercontent.com/analogdevicesinc/plutosdr-fw/master/scripts/53-adi-plutosdr-usb.rules

worked - but do we are going to be having lots of various udev rules/devices - do we want to put them in one place, or in separate repos?

Either works - it's just documentation that needs to be clear.

pcercuei, 2016-11-11:
Libiio doesn't care about VIDs and PIDs, it just looks for the "IIO" string. So to me it doesn't make sense to hardcode specific VIDs / PIDs in a udev rule shipped in the libiio repo.

@lclausen-adi's solution is much cleaner and would work for every libiio-compatible hardware. I don't think the rule is too broad.

pcercuei, 2016-12-14:
My own udev rule:
SUBSYSTEM=="usb", PROGRAM=="/bin/sh -c 'iio_info -s | grep %s{idVendor}:%s{idProduct}'", RESULT!="", GROUP="iio"

rgetz, 2017-01-29:
Closing, since @pcercuei committed this 2f0533c

closed 2017-01-29 by rgetz