analogdevicesinc/libiio #62: Question : identification of USB devices?

rgetz, 2016-11-14:
Is there a way to expose device USB serial number via libiio?
http://libusb.org/static/api-1.0/structlibusb__device__descriptor.html

I know I can manage things with uri (which port things are plugged into), but if I write an application that always expects device A to be set to frequency Z, and device B to be set to frequency Y, on USB, I can't really tell which is which. 

I have to ensure that end users always plug USB devices into the same ports. 

Or is there a way to get the USB serial number?

Thanks

pcercuei, 2016-11-15:
Michael's solution is here: https://github.com/analogdevicesinc/libiio/commit/0ef72e111ae0f7b914ad3e110b977dd3d1317c06
It adds the USB serial number to the backend description string of iio_context_info.

rgetz, 2016-11-15:
Ok - consider closed. Thanks

closed 2016-11-15 by rgetz