analogdevicesinc/libiio #68: examples don't build

rgetz, 2016-11-29:
analog@imhotep:~/github/libiio$ git show .
commit bb70e40f3026e99bdc60c50a96d49fe92f576869

analog@imhotep:~/github/libiio$ cd tests/
[CMakeOutput.log.txt](https://github.com/analogdevicesinc/libiio/files/619342/CMakeOutput.log.txt)

analog@imhotep:~/github/libiio/tests$ mkdir build
analog@imhotep:~/github/libiio/tests$ cd build/
analog@imhotep:~/github/libiio/tests/build$ cmake ..
-- The C compiler identification is GNU 6.2.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
CMake Error at CMakeLists.txt:39 (install):
  install TARGETS given no RUNTIME DESTINATION for executable target
  "iio_genxml".

-- Configuring incomplete, errors occurred!
See also "/home/analog/github/libiio/tests/build/CMakeFiles/CMakeOutput.log".
(attached)

rgetz, 2016-11-29:
doing this works...

analog@imhotep:~/github/libiio/build$cmake -DWITH_TESTS=1 ..

Is this the only way to build the tests?

pcercuei, 2016-11-29:
They are (and have always been) built by default along with the library.

rgetz, 2016-11-29:
Ok - maybe I just have never noticed them before. I will double check.

closed 2016-11-29 by rgetz