analogdevicesinc/libiio #70: iiod: dropping buffer refills

jcn9k, 2016-11-30:
I have a iio ADC driver using DMA to transfers.
When the userspace iio software is running on our embedded system using the local interface everything looks good. However using the network interface via iiod results in missing buffers, we are losing about a third of all data. Data is being read using iio_buffer_refill.

I'm digging into it to see if I can find the problem, but I've added some debug stated to iiod and can see iiod refilling at roughly the correct rate. But it is not issuing an writefd_io op for every refill. Sample rate is 24000sps with 4bytes per sample with a single channel.

Below is a log with some the additional messages inserted and timestamps added.

If anyone has any pointers on where to look that'd help.

Thanks
Phil

1480414079.731135920 DEBUG: Mask[0] = 0x00000001
1480414079.731309450 DEBUG: writefd_io: 2
1480414079.733963770 DEBUG: Waiting for completion...
1480414081.280262490 DEBUG: iio_buffer_refill: 40000
1480414081.280377950 DEBUG: writefd_io: 6
1480414081.280414930 DEBUG: writefd_io: 9
1480414081.280777680 DEBUG: writefd_io: 27731
1480414081.292457200 DEBUG: writefd_io: 12269
1480414081.296015000 DEBUG: Exiting rw_buffer with code 0
1480414081.301273900 DEBUG: Waiting for completion...
1480414081.974149370 DEBUG: iio_buffer_refill: 40000
1480414081.974287560 DEBUG: writefd_io: 6
1480414081.974333820 DEBUG: writefd_io: 9
1480414081.974646960 DEBUG: writefd_io: 36491
1480414081.976232780 DEBUG: writefd_io: 3509
1480414081.976325270 DEBUG: Exiting rw_buffer with code 0
1480414081.976500800 DEBUG: iio_buffer_refill: 40000
1480414081.982892750 DEBUG: Waiting for completion...
1480414082.656784750 DEBUG: iio_buffer_refill: 40000
1480414082.656931800 DEBUG: writefd_io: 6
1480414082.656974310 DEBUG: writefd_io: 9
1480414082.657551100 DEBUG: writefd_io: 39411
1480414082.658288270 DEBUG: writefd_io: 589
1480414082.658591230 DEBUG: Exiting rw_buffer with code 0
1480414082.669794630 DEBUG: Waiting for completion...
1480414083.339325110 DEBUG: iio_buffer_refill: 40000
1480414083.339503590 DEBUG: writefd_io: 6
1480414083.339584880 DEBUG: writefd_io: 9
1480414083.340001230 DEBUG: writefd_io: 40000
1480414083.340255160 DEBUG: Exiting rw_buffer with code 0
1480414083.340311110 DEBUG: iio_buffer_refill: 40000
1480414083.346527180 DEBUG: Waiting for completion...
1480414084.021636260 DEBUG: iio_buffer_refill: 40000
1480414084.021750630 DEBUG: writefd_io: 6
1480414084.021787380 DEBUG: writefd_io: 9
1480414084.022201650 DEBUG: writefd_io: 40000
1480414084.022271250 DEBUG: Exiting rw_buffer with code 0
1480414084.022478610 DEBUG: iio_buffer_refill: 40000
1480414084.028074920 DEBUG: Waiting for completion...
1480414084.704703950 DEBUG: iio_buffer_refill: 40000
1480414084.704878260 DEBUG: writefd_io: 6
1480414084.704958600 DEBUG: writefd_io: 9
1480414084.705379680 DEBUG: writefd_io: 40000
1480414084.707134320 DEBUG: Exiting rw_buffer with code 0
1480414084.716438780 DEBUG: Waiting for completion...
1480414085.387262830 DEBUG: iio_buffer_refill: 40000
1480414085.387453850 DEBUG: writefd_io: 6
1480414085.387531730 DEBUG: writefd_io: 9
1480414085.388030340 DEBUG: writefd_io: 40000
1480414085.388278010 DEBUG: Exiting rw_buffer with code 0
1480414085.388335590 DEBUG: iio_buffer_refill: 40000
1480414085.395464210 DEBUG: Waiting for completion...
1480414086.069969010 DEBUG: iio_buffer_refill: 40000
1480414086.070155530 DEBUG: writefd_io: 6
1480414086.070244510 DEBUG: writefd_io: 9
1480414086.070677660 DEBUG: writefd_io: 40000
1480414086.070921860 DEBUG: Exiting rw_buffer with code 0
1480414086.071049370 DEBUG: iio_buffer_refill: 40000
[... 134 lines cut ...]
1480414100.405654870 DEBUG: writefd_io: 6
1480414100.405693610 DEBUG: writefd_io: 9
1480414100.406093070 DEBUG: writefd_io: 40000
1480414100.406323900 DEBUG: iio_buffer_refill: 40000
1480414100.406649530 DEBUG: Exiting rw_buffer with code 0
1480414100.412089390 DEBUG: Waiting for completion...
1480414101.088227290 DEBUG: iio_buffer_refill: 40000
1480414101.088336770 DEBUG: writefd_io: 6
1480414101.088374430 DEBUG: writefd_io: 9
1480414101.088741900 DEBUG: writefd_io: 40000
1480414101.088966870 DEBUG: iio_buffer_refill: 40000
1480414101.089650050 DEBUG: Exiting rw_buffer with code 0
1480414101.097322430 DEBUG: Waiting for completion...
1480414101.770986160 DEBUG: iio_buffer_refill: 40000
1480414101.771098320 DEBUG: writefd_io: 6
1480414101.771135770 DEBUG: writefd_io: 9
1480414101.771926810 DEBUG: writefd_io: 40000
1480414101.776741350 DEBUG: Exiting rw_buffer with code 0
1480414101.793312100 DEBUG: Waiting for completion...
1480414102.453826260 DEBUG: iio_buffer_refill: 40000
1480414102.454018010 DEBUG: writefd_io: 6
1480414102.454121590 DEBUG: writefd_io: 9
1480414102.454570730 DEBUG: writefd_io: 40000
1480414102.454709220 DEBUG: Exiting rw_buffer with code 0
1480414102.454983960 DEBUG: iio_buffer_refill: 40000

pcercuei, 2016-11-30:
Hi,

Could you remove [this line](https://github.com/analogdevicesinc/libiio/blob/master/iiod/ops.c#L656) and see if it works better?

jcn9k, 2016-12-01:
G'day Paul,

That seems to work. Each refill now has a correspond write in my log.

Thanks
Phil

pcercuei, 2016-12-01:
Fixed it back in March, but it never made it to master. Thanks for the reminder!

closed 2016-12-01 by pcercuei