analogdevicesinc/libiio #72: stack usage

rgetz, 2016-11-30:
In most of the int iio_xxxxxx_attr_write_xxxxxx functions, we allocate a 1k buffer on the stack:

        char buf[1024];
        snprintf(buf, sizeof(buf), "%lld", val);

(In this case, of the longlong), we know at compile time, that the max length of long long is:

floor( log10( ULLONG_MAX ) ) + 1;

which should all be done by the pre-processor at complile time on modern toolsuites (it's all constants) - at least it works on gcc 6.2.0... which is:

floor ( log10(18446744073709551615) ) + 1 = 20;

which is alot more memory efficient than a 1k buffer on the stack. -- even DBL_MAX is only 309 characters...

?

pcercuei, 2016-12-01:
Hi,

So I don't really see this as a bug. We should have plenty of space on the stack and reserving 1024 bytes or 20 bytes doesn't make any difference speed-wise or memory-wise (we're talking about the stack, not the heap).

On the other hand, for really obvious cases like this I agree we could use saner defaults, not calculated at compile time (libiio must compile with really old & crappy compilers like Visual Studio).

rgetz, 2016-12-02:
Paul:

For memory constrained, embedded devices, 1k can be alot. Not everything has desktop memory.

If it better to calculate these by hand, and use that - I can do so (for 64-bit systems).

-Robin

rgetz, 2020-01-13:
with tinyiiod for lower end embedded systems, I am going to close this.

-robin

closed 2020-01-13 by rgetz