analogdevicesinc/libiio #73: `chan->extend_name` does not seem to be parsed correctly anymore

rlefevre, 2016-12-15:
Hi, 

It seems to me that `chan->extend_name` is not parsed anymore correctly for some time.

1. `get_short_attr_name()` does not handle it, therefore `handle_protected_scan_element_attr()` is not able to detect protected attributes, leading to an error
2. as protected attributes do not work when an `extend_name` is there, `set_channel_name()` is not able anymore to find the name and fix attributes (maybe unless the channel has 2 or more non protected attributes)

I'm not sure about the best way to fix this. Maybe `set_channel_name()` should be removed and a `get_channel_name()` function should be called at the same time than `get_channel_id()`, but I'm not sure yet we have enough information at this moment.

Did I miss something?

Thank you

pcercuei, 2016-12-16:
So channel names look to be parsed correctly here.
Could you give an example of a filename that is not parsed correctly? Or better, a XML of your context (generated with iio_genxml).

rlefevre, 2016-12-19:
One example of a failing channel is `scan_elements/in_current5_errm2_en`. It is created in the kernel by a custom driver filling `chan->extend_name` with `errm2`.

Here is some logs running `iio_genxml` with some debug added in `handle_protected_scan_element_attr()`:
```
DEBUG: Added channel 'timestamp' to device 'iio:device2'
DEBUG: handle_protected_scan_element_attr chd->id=current5 name=errm2_en path=scan_elements/in_current5_errm2_en
ERROR: handle_protected_scan_element_attr returns -22 (Invalid argument)
Unable to create IIO context
```
The error is fatal. Also even modifying `handle_protected_scan_element_attr()` is not enough as the channel is not renamed after (as explained in my previous message), so `iio_device_find_channel()` fails when using the `extend_name`.

Here is the debug output from an older version of libiio (about 1 year ago) that works correctly:
```
DEBUG: Added attr 'errm2_en' to channel 'current5'
DEBUG: Added attr 'errm2_type' to channel 'current5'
DEBUG: Added attr 'errm2_index' to channel 'current5'
DEBUG: Setting name of channel current5 to errm2
```

Do you think that the driver declares incorrectly this channel?

pcercuei, 2016-12-19:
The driver is probably fine.

The problem here, as you might have noticed, is that those sysfs filenames are difficult to parse. This is due to the fact that it was not designed from the start to be parsable.

Short example: if you have one IIO device with just one sysfs file `in_voltage0_my_sample_rate`:
* maybe the attribute is called `my_sample_rate` and there's no extended name,
* maybe the attribute is called `sample_rate` and the extended name is `my`,
* maybe the attribute is called `rate` and the extended name is `my_sample`.

Libiio works around this by analyzing all the attributes that correspond to one channel (aka. that start with `in_voltage0` in my example), and use the longest common string prefix as the extended name. In my example, since there's only one attribute, libiio would deduce the 3rd case. If I add a new sysfs attribute `in_voltage0_my_frequency`, libiio will correctly detect that 'my' is the extended name.
This is what `set_channel_name()` is about.

As you can see, this does not seem to work when the files in the scan_elements folder have extended names. I think what broke it is 9d77ca0. Before, they were added as regular scan_elements attributes, without any special handling, before the extended name calculation. I don't want to revert this commit, but I'll work on something to get it to work correctly.

rlefevre, 2016-12-19:
Thank you, I agree that this is commit 9d77ca0914297b03358e206cf0aef7bf6f2829b3 that broke it. I also understand that the format does not make parsing easy, that's why I didn't see a trivial solution. Also `set_channel_name()` is a lot less effective with protected attributes.

I can test an eventual fix if you need to once you have something.

Thank you.

pcercuei, 2016-12-19:
Could you try https://github.com/analogdevicesinc/libiio/tree/fix-scan-elements?

rlefevre, 2016-12-19:
Yes, it seems to fix the issue correctly:

```
...
DEBUG: Add protected attr 'errm2_en' to channel 'current5'
DEBUG: Add protected attr 'errm2_type' to channel 'current5'
DEBUG: Add protected attr 'errm2_index' to channel 'current5'
DEBUG: Setting name of channel current5 to errm2
...
```

pcercuei, 2016-12-19:
Nice :)

I'll wait for the review of @lclausen-adi to merge it to master, he will be back in January.

rlefevre, 2016-12-19:
Ok, thank you.

commit a471761d (2017-01-09): Local backend: Fix extended name not handled in scan-element attrs

Since the scan-element attributes are now hidden, they wouldn't get added
to the iio_channel's list of attributes, and as such didn't take part in
the extended name calculation.

Worse, the current code would fail when the scan-element attributes had
an extended name in their filename.

This commit changes this behaviour, so that the scan-element attributes
are first added in a separate list of attributes (in iio_channel_pdata).
The set_channel_name() function will then use the two lists of
attributes for the extended name cal [... cut]
closed 2017-01-09 by pcercuei