analogdevicesinc/libiio #79: When building on Ubuntu precise, build errors

rgetz, 2016-12-28:
Ubuntu 12.04 (precise), includes 2:1.0.9 of libusb-1.0-0
http://packages.ubuntu.com/precise/libusb-1.0-0-dev

Debian 7 (wheezy) includes 2:1.0.11-1 of libusb-1.0-0 (and will have the same issue)
https://packages.debian.org/wheezy/libusb-1.0-0

This does not include the ''libusb_set_auto_detach_kernel_driver'', which was added in 
https://github.com/libusb/libusb/blob/4eca418ef50dcaf3f2fca11158410acd72e8e2e2/ChangeLog
2013-07-11: v1.0.16

To get things to compile without errors requires:
https://github.com/analogdevicesinc/libiio/commit/74d80a44878f407498dc279717ca6eee34d3730f

To get things to actually work, (I think) will require a little more - most likely something the reverse of this:
https://sourceforge.net/p/libusbx/mailman/message/31055437/

If we want to support 3 year old verions of libusb - is something to be discussed.
Ubuntu precise EOLs April 2017 (which would seem to be fine)
https://wiki.ubuntu.com/Releases
Debian 7 EOLs May 2018 (LTS) (is  little further out to abandon).
https://wiki.debian.org/DebianReleases

-Robin

pcercuei, 2017-01-09:
So I think your workaround is fine. Typically we won't have any kernel driver registered for the IIO devices appearing over USB anyway.

rgetz, 2017-01-09:
Ok - I will leave that in the travis-ci branch? or you want a separate pull request for that one?

pcercuei, 2017-01-10:
I can cherry-pick it manually, if you allow me to set your Signed-Off-By.

rgetz, 2017-01-10:
Sure - or I can add it.

rgetz, 2017-01-11:
I included this in #84, so I'm closing this.

closed 2017-01-11 by rgetz