analogdevicesinc/libiio #81: runtime error on ubuntu 15

rgetz, 2017-01-02:
Same device works on Windows and other Linux installs

picozed@picozed:~$ iio_info -n 192.168.2.1
Library version: 0.8 (git tag: 9838779)
Compiled with backends: local xml network usb
IIO context created with network backend.
Backend version: 0.8 (git tag: v0.8   )
Backend description string: 192.168.2.1 Linux pluto 4.6.0-g1470706-dirty #66 SMP PREEMPT Fri Dec 23 12:40:07 CET 2016 armv7l
IIO context has 1 attributes:
	local,kernel: 4.6.0-g1470706-dirty
IIO context has 5 devices:
	iio:device0: adm1177
... works ....

picozed@picozed:~$ lsusb
Bus 001 Device 002: ID 1bcf:288a Sunplus Innovation Technology Inc. 
Bus 001 Device 006: ID 0456:b673 Analog Devices, Inc. 
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 005 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 004 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 003 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 002 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub

device is there

[ 2083.447606] usb 1-1: new high-speed USB device number 6 using ehci-pci
[ 2083.582909] usb 1-1: New USB device found, idVendor=0456, idProduct=b673
[ 2083.582935] usb 1-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 2083.582952] usb 1-1: Product: PlutoSDR (ADALM-PLUTO)
[ 2083.582967] usb 1-1: Manufacturer: Analog Devices Inc.
[ 2083.582982] usb 1-1: SerialNumber: 100000235523730700230029090216ea7c
[ 2083.590405] rndis_host 1-1:1.0 eth0: register 'rndis_host' at usb-0000:00:1d.7-1, RNDIS device, 00:e0:22:b6:0f:8b
[ 2083.591828] usb-storage 1-1:1.2: USB Mass Storage device detected
[ 2083.592234] scsi host7: usb-storage 1-1:1.2
[ 2083.593623] cdc_acm 1-1:1.3: ttyACM0: USB ACM device
[ 2083.668834] rndis_host 1-1:1.0 enx00e022b60f8b: renamed from eth0
[ 2083.729681] IPv6: ADDRCONF(NETDEV_UP): enx00e022b60f8b: link is not ready
[ 2084.593228] scsi 7:0:0:0: Direct-Access     Linux    File-Stor Gadget 0406 PQ: 0 ANSI: 2
[ 2084.594848] sd 7:0:0:0: Attached scsi generic sg1 type 0
[ 2084.596870] sd 7:0:0:0: [sdb] 32769 512-byte logical blocks: (16.7 MB/16.0 MiB)
[ 2084.597986] sd 7:0:0:0: [sdb] Write Protect is off
[ 2084.598009] sd 7:0:0:0: [sdb] Mode Sense: 0f 00 00 00
[ 2084.598847] sd 7:0:0:0: [sdb] Write cache: enabled, read cache: enabled, doesn't support DPO or FUA
[ 2084.612208]  sdb: sdb1
[ 2084.615682] sd 7:0:0:0: [sdb] Attached SCSI removable disk

Looks like everything is working ok.

picozed@picozed:~$ iio_info -s
Library version: 0.8 (git tag: 9838779)
Compiled with backends: local xml network usb
No contexts found.

but no devices found.

picozed@picozed:~$ dpkg -l libusb*
Desired=Unknown/Install/Remove/Purge/Hold
| Status=Not/Inst/Conf-files/Unpacked/halF-conf/Half-inst/trig-aWait/Trig-pend
|/ Err?=(none)/Reinst-required (Status,Err: uppercase=bad)
||/ Name                     Version           Architecture      Description
+++-========================-=================-=================-======================================================
ii  libusb-0.1-4:i386        2:0.1.12-27       i386              userspace USB programming library
ii  libusb-1.0-0:i386        2:1.0.19-1        i386              userspace USB programming library
ii  libusb-1.0-0-dev:i386    2:1.0.19-1        i386              userspace USB programming library development files

lclausen-adi, 2017-01-02:
Can you report the output of `uname -a` and ``file `which iio_info` ``

Thanks

rgetz, 2017-01-02:
picozed@picozed:~/github$ uname -a
Linux picozed 4.2.0-16-generic #19-Ubuntu SMP Thu Oct 8 14:46:51 UTC 2015 i686 i686 i686 GNU/Linux

picozed@picozed:~/github$ lsb_release -a
No LSB modules are available.
Distributor ID:	Ubuntu
Description:	Ubuntu 15.10
Release:	15.10
Codename:	wily

picozed@picozed:~/github$ which iio_info
/usr/bin/iio_info
picozed@picozed:~/github$ ls -l /usr/bin/iio_info 
-rwxr-xr-x 1 root root 31104 Jan  2 10:09 /usr/bin/iio_info

strace says it is opening /usr/lib/i386-linux-gnu/libiio.so.0

[strace_out.txt](https://github.com/analogdevicesinc/libiio/files/681005/strace_out.txt)

picozed@picozed:~/github$ ls -l /usr/lib/i386-linux-gnu/libiio.so.0
lrwxrwxrwx 1 root root 13 Jan  2 10:15 /usr/lib/i386-linux-gnu/libiio.so.0 -> libiio.so.0.8
picozed@picozed:~/github$ ls -l /usr/lib/i386-linux-gnu/libiio.so.0.8 
-rw-r--r-- 1 root root 337504 Jan  2 10:09 /usr/lib/i386-linux-gnu/libiio.so.0.8

lclausen-adi, 2017-01-02:
Can you attach (or send by email) the full strace output of `iio_info --scan`.

Thanks.

rgetz, 2017-01-02:
That should have been what was above (I think)

picozed@picozed:~/github$ strace -ostrace_out.txt iio_info --scan
Library version: 0.8 (git tag: 9838779)
Compiled with backends: local xml network usb
No contexts found.

[strace_out.txt](https://github.com/analogdevicesinc/libiio/files/681113/strace_out.txt)

ltrace was a lot less interesting.

picozed@picozed:~/github$ ltrace -o ltrace_out.txt iio_info --scan
Library version: 0.8 (git tag: 9838779)
Compiled with backends: local xml network usb
No contexts found.

[ltrace_out.txt](https://github.com/analogdevicesinc/libiio/files/681116/ltrace_out.txt)

lclausen-adi, 2017-01-02:
Sorry, didn't see the file.

open("/dev/bus/usb/001/002", O_RDWR)    = -1 EACCES (Permission denied)
**open("/dev/bus/usb/001/006", O_RDWR)    = -1 EACCES (Permission denied)**
open("/dev/bus/usb/001/001", O_RDWR)    = -1 EACCES (Permission denied)
open("/dev/bus/usb/005/001", O_RDWR)    = -1 EACCES (Permission denied)
open("/dev/bus/usb/004/001", O_RDWR)    = -1 EACCES (Permission denied)
open("/dev/bus/usb/003/001", O_RDWR)    = -1 EACCES (Permission denied)
open("/dev/bus/usb/002/001", O_RDWR)    = -1 EACCES (Permission denied)

Looks like the user has no access to the USB device.

rgetz, 2017-01-02:
Ok - face palm - I didn't remember to install the udev rules. #57 

Is there a way to have iio_info say permission error, or some meaningful error?

but I still get the same error.

picozed@picozed:~/github$ sudo strace -ostrace_out.txt iio_info -s
Library version: 0.8 (git tag: 9838779)
Compiled with backends: local xml network usb
No contexts found.
picozed@picozed:~/github$ grep dev/bus strace_out.txt 
openat(AT_FDCWD, "/dev/bus/usb", O_RDONLY|O_NONBLOCK|O_LARGEFILE|O_DIRECTORY|O_CLOEXEC) = 3
open("/dev/bus/usb/001/002", O_RDWR)    = 11
open("/dev/bus/usb/001/001", O_RDWR)    = 11
open("/dev/bus/usb/005/001", O_RDWR)    = 11
open("/dev/bus/usb/004/001", O_RDWR)    = 11
open("/dev/bus/usb/003/001", O_RDWR)    = 11
open("/dev/bus/usb/002/001", O_RDWR)    = 11

[strace_out.txt](https://github.com/analogdevicesinc/libiio/files/681127/strace_out.txt)

rgetz, 2017-01-02:
nevermind - the last time I tried - it wasn't plugged in (the USB had fallen out).

rgetz, 2017-01-02:
I will close this - and send a patch for iio_info / permission later.

closed 2017-01-02 by rgetz

lclausen-adi, 2017-01-02:
The problem is without being able to access the device we won't be able to check if the device is a IIO device or not. And we don't want to report an permission error for every USB device in the system that can't be accessed, since that would be quite a lot.

rgetz, 2017-01-02:
What I was working on is keeping track, reporting a permission error (from a scan) if you can't see them all (and only if not root).

The error string would be something like "Got permission errors on all USB devices, maybe udev error?" 

Thoughts?

lclausen-adi, 2017-01-02:
well, you'd get that message as well if no IIO devices are plugged in. Not sure if this is what we want. A more reliable way to detect this would be nice.

rgetz, 2017-01-02:
Yeah, this is where I am now. All permission errors when no devices are plugged in, and running as normal user. 

When running as root - you see devices, and can tell from that.

rgetz, 2017-01-03:
OK - where I'm at.

You can use libusb to find the pid/vid, without opening the device (libusb_get_device_descriptor(dev, &desc); works, even when libusb_open(dev, &hdl) fails with LIBUSB_ERROR_ACCESS). However - the interface description isn't in the device descriptor, and like was discussed in the past - I don't think we want to put pid/vid tables into libiio. (although that would solve the problem).
http://libusb.org/static/api-1.0/structlibusb__device__descriptor.html
The interface - is in the interface descriptor, which needs the device handle (which you get from libusb_open, which fails in this case).
http://libusb.sourceforge.net/api-1.0/structlibusb__interface__descriptor.html
so we can't use that.

I did notice, that even for a normal user "IIO" shows up in sysfs as readable.

cat /sys/bus/usb/devices/3-2:1.5/interface
IIO

but didn't want to go trolling around sysfs if I didn't have to (and thought that libusb should handle this for me) - but have not dug into it far enough yet.

-Robin