analogdevicesinc/libiio #85: Current CMake doesn't build tarball on os-x

rgetz, 2017-01-12:
On an OS-X machine:

cd build; rm -rf *; cmake ..; make ; make package; ls -lG ; tar tf libiio-0.8.gc59cba6-Darwin.tar.gz

gives me something like:
-rw-r--r--   1 analogdevices  staff     29 Jan 12 14:03 libiio-0.8.gc59cba6-Darwin.tar.gz

which is an empty tar file. :(

making this change:
--- a/CMakeLists.txt
+++ b/CMakeLists.txt
@@ -29,7 +29,7 @@ set(BUILD_SHARED_LIBS ON CACHE BOOL "Build shared libraries")
 
 if(${CMAKE_SYSTEM_NAME} MATCHES "Darwin")
        set(CMAKE_MACOSX_RPATH ON)
-       set(SKIP_INSTALL_ALL ON)
+#      set(SKIP_INSTALL_ALL ON)
 endif()

and doing it again, makes a tar file with everything in it. (but I'm sure this breaks "make install", but that might not matter since we now have a pkg).

 -rw-r--r--   1 analogdevices  staff  58314 Jan 12 14:06 libiio-0.8.gc59cba6-Darwin.tar.gz
bin/
bin/iio_adi_xflow_check
bin/iio_genxml
bin/iio_info
bin/iio_readdev
bin/iio_reg
lib/
lib/iio.framework/
lib/iio.framework/Headers
lib/iio.framework/iio
lib/iio.framework/Resources
lib/iio.framework/Versions/
lib/iio.framework/Versions/0.8/
lib/iio.framework/Versions/0.8/Headers/
lib/iio.framework/Versions/0.8/Headers/iio.h
lib/iio.framework/Versions/0.8/iio
lib/iio.framework/Versions/0.8/Resources/
lib/iio.framework/Versions/0.8/Resources/Info.plist
lib/iio.framework/Versions/Current
lib/pkgconfig/
lib/pkgconfig/libiio.pc

closed 2017-01-18 by pcercuei

pcercuei, 2017-01-18:
So I can't make it generate both a tarball and the .pkg.
But now you can pass the option OSX_PACKAGE (ON by default) to generate the .pkg, and disable it when you want to create the tarball.

rgetz, 2017-01-18:
Sounds great. That should work. I will need to tweak the travis files - but I needed to do that anyway (based on our phone discussion).

-Robin

commit 6e3fa6fc (2017-01-17): CMake: Add an option to disable creation of OSX package

The OSX_PACKAGE option, ON by default, can be unset to disable the
creation of a OSX package.

Fixes #85.