analogdevicesinc/libiio #89: Macro definition of snprintf conflicts on MSVC 14 (VS2015)

mfumita, 2017-01-23:
There is the following code in "iio-private.h".

```C++
#ifdef _MSC_BUILD
#define snprintf sprintf_s  
#define inline __inline  
#endif
```

This cases a build error in Microsoft Visual C++ version 14 which comes from Visual Studio 2015 because "snprintf" is already defined as below.

> Error C1189 #error:  Macro definition of snprintf conflicts with Standard Library function declaration

[https://msdn.microsoft.com/en-us/library/2ts7cx93.aspx](https://msdn.microsoft.com/en-us/library/2ts7cx93.aspx)

I would like to suggest to change the code like below.
```C++
#if _MSC_VER<=1800
#define snprintf sprintf_s
#define inline __inline
#endif
```

pcercuei, 2017-01-23:
But then MSVC will complain that we're using snprintf( ) and not the secure version snprintf_s( )...
I think the best way would be to have a iio_snprintf( ) macro that resolves either to snprintf_s if present, or snprintf otherwise.

mfumita, 2017-01-24:
Thank you for your reply.

I think having a iio_snprintf( ) macro is a proper way as you said. It would be great if you would make this change once you have a chance.

commit e53ae634 (2017-01-24): Use macro iio_snprintf instead of snprintf

The problem of using snprintf directly is that Visual Studio will
complain that we're not using the more secure sprintf_s variant.

The way to work around that has been to redefine 'snprintf' as a macro
set to 'sprintf_s', so that all calls in the code to snprintf would
resolve as calls to the sprintf_s macro.

However, Visual Studio 2015 complains that we're shadowing a standard
library function. The solution here, is to change all calls to snprintf
in the code to a macro iio_snprintf that we can point either to
sprintf_s if available, or snprintf.
 [... cut]
commit ca799054 (2017-01-24): Use macro iio_snprintf instead of snprintf

The problem of using snprintf directly is that Visual Studio will
complain that we're not using the more secure sprintf_s variant.

The way to work around that has been to redefine 'snprintf' as a macro
set to 'sprintf_s', so that all calls in the code to snprintf would
resolve as calls to the sprintf_s macro.

However, Visual Studio 2015 complains that we're shadowing a standard
library function. The solution here, is to change all calls to snprintf
in the code to a macro iio_snprintf that we can point either to
sprintf_s if available, or snprintf.
 [... cut]
commit a2d460df (2017-01-24): Use macro iio_snprintf instead of snprintf

The problem of using snprintf directly is that Visual Studio will
complain that we're not using the more secure sprintf_s variant.

The way to work around that has been to redefine 'snprintf' as a macro
set to 'sprintf_s', so that all calls in the code to snprintf would
resolve as calls to the sprintf_s macro.

However, Visual Studio 2015 complains that we're shadowing a standard
library function. The solution here, is to change all calls to snprintf
in the code to a macro iio_snprintf that we can point either to
sprintf_s if available, or snprintf.
 [... cut]
closed 2017-01-24 by pcercuei