analogdevicesinc/libiio #90: all the language bindings don't get installed in all packages

rgetz, 2017-01-25:
After I build, the header files are include in the packges:

analog@imhotep:~/github/libiio$ find ./ -name "iio.h"
./iio.h
./build/_CPack_Packages/Linux/DEB/libiio-0.8.gc4040b6-Linux/usr/include/iio.h
./build/_CPack_Packages/Linux/RPM/libiio-0.8.gc4040b6-Linux/usr/include/iio.h
./build/_CPack_Packages/Linux/TGZ/libiio-0.8.gc4040b6-Linux/include/iio.h

in the deb, rpm, and tar file.

However:
analog@imhotep:~/github/libiio$ find ./ -name iio.py
./bindings/python/iio.py
./build/_CPack_Packages/Linux/DEB/libiio-0.8.gc4040b6-Linux/usr/lib/python2.7/site-packages/iio.py
./build/_CPack_Packages/Linux/RPM/libiio-0.8.gc4040b6-Linux/usr/lib/python2.7/site-packages/iio.py
./build/bindings/python/build/lib.linux-x86_64-2.7/iio.py

in the deb, rpm, but not the tar file.

analog@imhotep:~/github/libiio$ find ./ -name "iio-wrapper.h"
./bindings/matlab/iio-wrapper.h

not in the deb, rpm, or tar file.

pcercuei, 2017-01-27:
Fixed in latest master by commit abd9caf.

rgetz, 2017-01-31:
Are we sure this works? I just looked in the 0.9 tarball, and there is no iio-wrapper.h (the python files are there, but no MATLAB bindings).

tar tf libiio-0.9.g9da4004-Linux.tar.gz | grep wrapper

nothing.

pcercuei, 2017-02-01:
One question, the iio-wrapper.h should only be installed in the Linux/OSX tarballs, not in the DEB/RPM, right?

rgetz, 2017-02-01:
I think it is all. 

Any user, who wants to use MATLAB, needs the wrapper. It shouldn't matter what package they install.

pcercuei, 2017-02-01:
Didn't the Matlab guys say that they only need the windows ZIP and OSX/Linux tarballs?

rgetz, 2017-02-01:
From an end user perspective - we shouldn't force people who want to use iio/MATLAB to install from a specific install method (tarball) - should we?

pcercuei, 2017-02-01:
Well, where should the Matlab bindings be installed then?

rgetz, 2017-02-02:
I'm not 100% sure ...

@mfornero : Where do you want things?

emsozer, 2017-02-03:
The support package installer will use tarball. We do not have plans to use deb or rpm at this point.

rgetz, 2017-02-03:
@emsozer - that really doesn't answer the question. Where in the tarball do you want things?

I understand the support package will not use the deb or rpm, but other users may want to for their own projects. There needs to be a common place for everything, where the documentation points to end users.

mfornero, 2017-02-03:
I *think* we can tolerate the location being anywhere inside the install directory, as long as we know where it is. I believe we specify the full path to both iio-wrapper.h and iio.h as part of the loadlibrary command.

We may also end up putting both files on the MATLAB path (I can't recall if this was necessary for loadlibrary to work correctly)-- in which case we can simply add the folder of iio.h as well as iio-wrapper.h

However, if we are adding the folder that contains iio-wrapper.h to the MATLAB path, we'd also end up picking up all of the MATLAB bindings. At some point in the future, it may be useful to place the system object and other files inside of a MATLAB package (i.e. inside a +libiio directory), but I think we can defer that for now.

emsozer, 2017-02-03:
I agree with @mfornero. As long as we know where the header files are, we can handle it on the MATLAB side (basically add those locations to MATLAB path). It helps if both iio.h and iio-wrapper.h are in the same directory but not essential.

rgetz, 2017-02-08:
OK - I think this is resolved now (mostly) - it's in the tar files, but not the deb or the rpm.

closed 2017-02-08 by rgetz