analogdevicesinc/libiio #95: Don’t try to build the C# bindings using the Chicken Scheme compiler

linkmauve, 2017-02-10:
Both .NET and Chicken Scheme seem to provide a binary named csc, it would be helpful to test which one it actually is during the cmake step, instead of trying to compile C# code as Scheme.

This broke the ArchLinux package of libiio I maintain, for now I did workaround that by disabling the .NET bindings.

rgetz, 2017-02-10:
Thanks for reporting. 

Do you have a suggested patch?

linkmauve, 2017-02-10:
I haven’t had a look yet, I think the ideal thing would be for cmake or extra-cmake-modules to provide a C# module.

I found this example, but it does seem to be quite Windows-specific: https://github.com/SimpleITK/SimpleITK/blob/master/CMake/FindCSharp.cmake

pcercuei, 2017-02-11:
I think csc is only for compiling the bindings on Windows; Mono's C# compiler is called msc.
So I could just adapt the CMake script to look for the right name on each OS.

commit b85f83cd (2017-02-14): CMake: Only use 'csc' as the C# compiler on Windows

On Linux, 'csc' may also be the name of the Chicken Scheme Compiler.

Fixes #95.
closed 2017-02-14 by pcercuei