analogdevicesinc/libiio #98: man pages for applications in ./tests

rgetz, 2017-05-02:
should we build man pages for iio_info, iio_readdev, and iio_reg?

or just put things in the wiki? (and the man pages point to the wiki?)

-Robin

pcercuei, 2017-05-03:
It would be great to have man pages, yes.

rgetz, 2017-05-30:
What should be the master format?

http://asciidoc.org/a2x.1.html
seems to be the way many other cmake projects do it.

(rather than just creating actual troff files).

?

pcercuei, 2017-06-01:
a2x uses asciidoc, which is itself a 1.0 GiB download size / 1.5 GiB install size on my Debian if I try to apt-get it...
I'm more in favour of not adding dependencies, so I'd prefer actual TROFF files.

rgetz, 2017-06-01:
OK - closing this, then, expect something tomorrow.

closed 2017-06-01 by rgetz