analogdevicesinc/libiio #101: CMake: Could NOT find LibLZMA

lutherling12, 2017-06-08:
I was cross compiling libiio and came across this error message from CMake.
```
Could NOT find LibLZMA (missing: LIBLZMA_HAS_AUTO_DECODER
LIBLZMA_HAS_EASY_ENCODER LIBLZMA_HAS_LZMA_PRESET) (found version "5.3.0")
```
So... LibLZMA can't be found, but version 5.3.0 has been found... Some quick googling from [CMake documentation of my version of CMake](https://cmake.org/cmake/help/v3.0/module/FindLibLZMA.html) yields

> LIBLZMA_HAS_AUTO_DECODER  - True if lzma_auto_decoder() is found (required).
> LIBLZMA_HAS_EASY_ENCODER  - True if lzma_easy_encoder() is found (required).
> LIBLZMA_HAS_LZMA_PRESET   - True if lzma_lzma_preset() is found (required).

Scrolling up a bit on the output, the path of the shared object is given:

```
-- Looking for lzma_auto_decoder in <prefix>/lib/liblzma.so
-- Looking for lzma_auto_decoder in <prefix>/lib/liblzma.so - not found
-- Looking for lzma_easy_encoder in <prefix>/lib/liblzma.so
-- Looking for lzma_easy_encoder in <prefix>/lib/liblzma.so - not found
-- Looking for lzma_lzma_preset in <prefix>/lib/liblzma.so
-- Looking for lzma_lzma_preset in <prefix>/lib/liblzma.so - not found
```

Using nm -D on shared object located at <same prefix>/lib/liblzma.so provides the following:

```
                 U calloc
                 U clock_gettime
                 w __cxa_finalize
                 U free
                 w __gmon_start__
                 w _ITM_deregisterTMCloneTable
                 w _ITM_registerTMCloneTable
                 w _Jv_RegisterClasses
00000000000099e0 T lzma_alone_decoder
0000000000005760 T lzma_alone_encoder
---> 0000000000009d30 T lzma_auto_decoder <---
0000000000005d20 T lzma_block_buffer_bound
0000000000009db0 T lzma_block_buffer_decode
0000000000005d30 T lzma_block_buffer_encode
0000000000003c10 T lzma_block_compressed_size
000000000000a340 T lzma_block_decoder
0000000000006150 T lzma_block_encoder
000000000000a400 T lzma_block_header_decode
00000000000062c0 T lzma_block_header_encode
00000000000061c0 T lzma_block_header_size
0000000000003c70 T lzma_block_total_size
0000000000005d80 T lzma_block_uncomp_encode
0000000000003b80 T lzma_block_unpadded_size
000000000000ce00 T lzma_check_is_supported
000000000000ce20 T lzma_check_size
00000000000036e0 T lzma_code
00000000000054c0 T lzma_cputhreads
000000000000cf10 T lzma_crc32
000000000000d040 T lzma_crc64
0000000000006440 T lzma_easy_buffer_encode
000000000000a600 T lzma_easy_decoder_memusage
---> 00000000000064f0 T lzma_easy_encoder <---
0000000000006570 T lzma_easy_encoder_memusage
00000000000039a0 T lzma_end
000000000000b100 T lzma_file_info_decoder
000000000000b370 T lzma_filter_decoder_is_supported
0000000000006770 T lzma_filter_encoder_is_supported
000000000000b4c0 T lzma_filter_flags_decode
0000000000006b40 T lzma_filter_flags_encode
0000000000006ad0 T lzma_filter_flags_size
0000000000003d90 T lzma_filters_copy
0000000000006870 T lzma_filters_update
0000000000003a10 T lzma_get_check
00000000000039e0 T lzma_get_progress
0000000000004a80 T lzma_index_append
0000000000004830 T lzma_index_block_count
000000000000bab0 T lzma_index_buffer_decode
0000000000007000 T lzma_index_buffer_encode
0000000000004d00 T lzma_index_cat
0000000000004940 T lzma_index_checks
000000000000ba30 T lzma_index_decoder
0000000000004f30 T lzma_index_dup
0000000000006f90 T lzma_index_encoder
0000000000004730 T lzma_index_end
00000000000048b0 T lzma_index_file_size
000000000000bdc0 T lzma_index_hash_append
000000000000be60 T lzma_index_hash_decode
000000000000bd90 T lzma_index_hash_end
000000000000bcc0 T lzma_index_hash_init
000000000000bda0 T lzma_index_hash_size
00000000000046d0 T lzma_index_init
0000000000005170 T lzma_index_iter_init
0000000000005310 T lzma_index_iter_locate
0000000000005180 T lzma_index_iter_next
0000000000005140 T lzma_index_iter_rewind
00000000000047a0 T lzma_index_memusage
0000000000004820 T lzma_index_memused
0000000000004850 T lzma_index_size
0000000000004840 T lzma_index_stream_count
0000000000004980 T lzma_index_stream_flags
0000000000004a00 T lzma_index_stream_padding
0000000000004880 T lzma_index_stream_size
0000000000004870 T lzma_index_total_size
0000000000004930 T lzma_index_uncompressed_size
---> 0000000000010560 T lzma_lzma_preset <---
0000000000003aa0 T lzma_memlimit_get
0000000000003b10 T lzma_memlimit_set
0000000000003a30 T lzma_memusage
000000000000edd0 T lzma_mf_is_supported
0000000000012280 T lzma_mode_is_supported
00000000000041d0 T lzma_physmem
000000000000b440 T lzma_properties_decode
0000000000006a60 T lzma_properties_encode
0000000000006a00 T lzma_properties_size
000000000000b190 T lzma_raw_buffer_decode
00000000000065e0 T lzma_raw_buffer_encode
000000000000b3c0 T lzma_raw_decoder
000000000000b430 T lzma_raw_decoder_memusage
00000000000067e0 T lzma_raw_encoder
0000000000006860 T lzma_raw_encoder_memusage
0000000000007120 T lzma_stream_buffer_bound
000000000000c210 T lzma_stream_buffer_decode
0000000000007160 T lzma_stream_buffer_encode
000000000000cb10 T lzma_stream_decoder
0000000000007a40 T lzma_stream_encoder
0000000000009440 T lzma_stream_encoder_mt
00000000000094c0 T lzma_stream_encoder_mt_memusage
0000000000005410 T lzma_stream_flags_compare
000000000000cc10 T lzma_stream_footer_decode
0000000000007b30 T lzma_stream_footer_encode
000000000000cb90 T lzma_stream_header_decode
0000000000007ad0 T lzma_stream_header_encode
0000000000003400 T lzma_version_number
0000000000003410 T lzma_version_string
000000000000cca0 T lzma_vli_decode
0000000000007bc0 T lzma_vli_encode
0000000000005490 T lzma_vli_size
                 U malloc
                 U memcmp
                 U memcpy
                 U memmove
                 U memset
                 U pthread_condattr_destroy
                 U pthread_condattr_init
                 U pthread_condattr_setclock
                 U pthread_cond_destroy
                 U pthread_cond_init
                 U pthread_cond_signal
                 U pthread_cond_timedwait
                 U pthread_cond_wait
                 U pthread_create
                 U pthread_join
                 U pthread_mutex_destroy
                 U pthread_mutex_init
                 U pthread_mutex_lock
                 U pthread_mutex_unlock
                 U pthread_sigmask
                 U __sched_cpucount
                 U sched_getaffinity
                 U sigfillset
                 U __stack_chk_fail
                 U sysconf
0000000000000000 A XZ_5.0
0000000000000000 A XZ_5.2
0000000000000000 A XZ_5.3.0alpha
```

All three functions appear. So am I maybe using the wrong version of LibLZMA?
I'm not sure if this is a CMake or Libiio issue at this point, but if I'm posting this issue in the wrong area, please let me know.

pcercuei, 2017-06-08:
So I don't know what you're doing, libiio doesn't use libLZMA :)

rgetz, 2017-09-27:
closing this - since I don't think this is a libiio issue. (as Paul stated, it's not used).

closed 2017-09-27 by rgetz