analogdevicesinc/libiio #102: examples/ad9361-iiostream.c failed on Windows, using release mode

josephsmeng, 2017-06-18:
examples/ad9361-iiostream.c  works right on Windows when using debug mode but failed in release mode build.

in release mode, iio_create_network_context return 0, but in debug mode libiio can create context successfully.  In both mode, the source code and hardware are the same.

the compiler we using is VS2017 64 bit mode, OS Windows 7.

rgetz, 2017-06-18:
Are you using the pre-built DLL to link to? Or building the entire thing from scratch?

https://appveyorcidata.blob.core.windows.net/analogdevicesinc-17855/libiio/2015-1-942/q7nji8p9sp8x7qnf/libiio-setup.exe?sv=2015-12-11&sr=c&sig=klC%2F77eQD3VX7DM%2FDJeb%2BksmhwMQBAuCzBoxfcIJ6DE%3D&st=2017-06-18T23%3A50%3A21Z&se=2017-06-18T23%3A56%3A21Z&sp=r

pcercuei, 2017-06-19:
My guess - VS2017 strips the assert( ) calls when compiled in Release mode.

Code like
```assert(ctx = iio_create_default_context() && "No context");```
is not only super-ugly but very error-prone.

I you just write instead:
```ctx = iio_create_default_context());```
I'm pretty sure it will fix the problem.

josephsmeng, 2017-06-19:
i don't have the hardware to test right now.
I tried pre-build DLL and build one from scratch, using 0.10 and the latest git
but i think @pcercuei  was right, maybe we should change the assert usage.

rgetz, 2017-06-19:
@pcercuei is normally correct - here are some quick pointers to look at (now I know what to search for):

https://stackoverflow.com/questions/270488/is-assertfalse-ignored-in-release-mode-vc

Quick test:
https://codeyarns.com/2011/05/30/visual-c-assert-in-release-mode/

mhennerich, 2017-07-19:
This issue should be fixed with this commit: 6e27fb403c070a1269926ef8ea4ed08570132eaf

pcercuei, 2017-07-19:
Yep. Closing this bug report then.

closed 2017-07-19 by pcercuei