analogdevicesinc/libiio #104: bug in dummy-iiostream.c not working with previous library versions

ycdtosa, 2017-06-23:
There is a check for older versions where fmt does not have a repeat.

```
	/* check for struct iio_data_format.repeat support */
	has_repeat = major >= 0 && minor >= 8 ? true : false;

```
but when has_repeat is false, precisely because iio_data_format.repeat is not supported
by that library version, the compiler still has to deal with....

```
static ssize_t sample_cb(const struct iio_channel *chn, void *src, size_t bytes, void *d)
{
	const struct iio_data_format *fmt = iio_channel_get_data_format(chn);
	unsigned int repeat = has_repeat ? fmt->repeat : 1;
        ...
}

```
and it can not because fmt->repeat does not exist.

I am afraid this has be addressed with macros instead.

of course, it does work when the library supports iio_channel_fmt.repeat

rgetz, 2020-01-13:
I think that this works on old kernels with new libraries. Since this is so old and no one else has complained, I will close with a won't fix. 

-Robin

closed 2020-01-13 by rgetz

ycdtosa, 2020-01-20:
Are you aware that under certain configurations the code does not compile.

this line:

`has_repeat ? fmt->repeat : 1
`

When `has_repeat` is false, because the compiler still needs to compile `fmt->repeat` which does not exist at that version.

Some preprocessor magic might be needed there. 

If it does not compile, the code to set `has_repeat` is not needed at all, since it only works if it does and every test checking that value is also unnecessary. Any `has_repeat` related code is not needed, since has_repeat is always true. In this case a message saying "versions before XXX are no longer supported" would be more helpful.

Cheers.