analogdevicesinc/libiio #106: quiet mode in iio_attr

rgetz, 2017-07-14:
Currently when asking the setting of one device, things are pretty noisy.

analog@imhotep:~$ iio_attr -u usb:3.80.5  -D loopback
dev 'ad9361-phy', debug attr 'loopback', value :'0'

If you just want the value, you need to get the last field, and strip off the single quotes.

@mhennerich asked for a flag (quiet) that would just do:

analog@imhotep:~$ iio_attr -u usb:3.80.5 -q  -D loopback
0

I think this was to help with a project he @adisuciu were working on.

I wanted to check to see if that was OK with @pcercuei and @larsclausen.

I assume that if you are doing a write:

analog@imhotep:~$ iio_attr -u usb:3.80.5 -q  -D loopback 1
1

the return value is the value after the write.

Thoughts/comments?

pcercuei, 2017-07-17:
Fine with me.

mhennerich, 2017-08-09:
submitted a pull request #111

closed 2017-09-18 by rgetz