analogdevicesinc/libiio #109: iio.Buffer(): returns error in Python

damercer, 2017-08-01:
This seems to be an actual bug in the Python bindings for libiio (when used with M2K at least).
When I run the attached test program it returns the following error:

Python 2.7.12 (v2.7.12:d33e0cf91556, Jun 27 2016, 15:19:22) [MSC v.1500 32 bit (Intel)] on win32
Type "copyright", "credits" or "license()" for more information.
>>>
RESTART: C:\Users\Doug\Documents\RPI\ADiscovery\ALM\libiio-test\test_m2k-1.py
Library version: 0.10 (git tag: c95ff6a)
Linux (none) 4.6.0-ge4025f7 #34 SMP PREEMPT Wed Jul 5 13:29:17 CEST 2017 armv7l
(0L, 10L, u'v0.10 ')
DAC1 readback: 0.000000
DAC2 readback: 0.000000
ADC0 readback: 1650.781250
ADC1 readback: 0.001563
ADC2 readback: 0.001563
ADC3 readback: 640.625000
ADC4 readback: 707.031250
ADC5 readback: 1388.281250
m2k_dac_a = <iio.Device object at 0x028F3490>
dac_a0 = <iio.Channel object at 0x028FB110>
dac_a0 enabled? True
m2k_dac_a buffer count = 0 <bound method Device.set_kernel_buffers_count of <iio.Device object at 0x028F3490>>
m2k_dac_a sample size = 0
m2k_fabric buffer count = 0 <bound method Device.set_kernel_buffers_count of <iio.Device object at 0x028E01B0>>

Traceback (most recent call last):
File "C:\Users\Doug\Documents\RPI\ADiscovery\ALM\libiio-test\test_m2k-1.py", line 87, in <module>
dac_a_buff = iio.Buffer(m2k_dac_a, 4096, True)
File "C:\Users\Doug\Documents\RPI\ADiscovery\ALM\libiio-test\iio.py", line 516, in __init__
self._buffer = _create_buffer(device._device, samples_count, cyclic)
File "C:\Users\Doug\Documents\RPI\ADiscovery\ALM\libiio-test\iio.py", line 33, in _checkNull
raise OSError(err, _strerror(err))
OSError: [Errno 6] No such device or address
>>>

Note: the dac_a0 channel is enabled before trying to create buffer for device m2k_dac_a.

Thanks

Doug
[test_m2k-1.zip](https://github.com/analogdevicesinc/libiio/files/1190709/test_m2k-1.zip)

lclausen-adi, 2017-08-01:
Hi,

The property is called 'enabled' and not 'enable'.

damercer, 2017-08-01:
Some examples that actually send info back and forth to the hardware ( 
more than what iio_info.py does ) would be a huge help. Could something 
along those lines be provided?

Thanks

Doug

mhennerich, 2017-08-01:
Here is a small example - I put together some time ago...

#!/usr/bin/env python

import matplotlib.pyplot as plt
import numpy as np
import math
import iio
import struct

def iio_sample ( ):
	buf.refill()
	return buf.read()

target_device = "AD7980-3x-sync"

temp_device = "adt7420"

N = 16384

ctx = iio.Context()

print ('Library version: %u.%u (git tag: %s)' % iio.version)
print ('IIO context created: ' + ctx.name)
print ('Backend version: %u.%u (git tag: %s)' % ctx.version)
print ('Backend description string: ' + ctx.description)
print ('IIO context has %u devices:' % len(ctx.devices))

for dev in ctx.devices:
    print '\t' + dev.name

dev = ctx.find_device(target_device)
temp = ctx.find_device(temp_device)

print '\n\tFound ' + dev.id + ': ' + dev.name

Fs = float(dev.channels[0].attrs["sampling_frequency"].value)
T = float(temp.attrs["in_temp_input"].value) / 1000

print "\tT = %f deg C" % T
print "\tFS = %f MHz" % (Fs / 1000000) 

window = np.hanning(N)
window = window / (2**15 - 1)

dev.channels[0].enabled  = False
dev.channels[1].enabled  = True
dev.channels[2].enabled  = False

buf = iio.Buffer(dev, N)

while True:
	x = iio_sample()
        X = []
	for n in range (0, len(x), 2):
		X.append(struct.unpack_from(">H", x, n)[0])

#	print X

#	plt.plot(np.arange(N), X);
#	plt.show()

        X = X * window 

	FFT = np.fft.fft(X) / N  * 4
	FFT = FFT[0:N/2]
	freq = np.arange(N/2) * Fs / (N * 1000000)

	plt.plot(freq, 20 * np.log10(np.abs(FFT)))
	plt.grid(False)
	plt.suptitle('FFT (%u points)' % N)
        plt.autoscale(False)
        plt.ylim([-140,0])
        plt.draw()
        plt.pause(0.01)
        plt.clf()

closed 2017-08-01 by damercer