analogdevicesinc/libiio #110: Writing buffers to M2K AWG DACs?

damercer, 2017-08-07:
I've been attempting to use the iio.py Python binding to write buffers to the AWG DACs in M2K. Here is my test program. I make a ramp of integers from 0 to 4095 ( or -2048 to 2047 ) and seem to need to convert the array to a numpy array to not produce an error, but the analog output does not produce anything like a ramp. It should be a cyclic ramp if I have done things correctly. There is a short burst of what looks like a square wave? I doubt I am formatting the data correctly but I would expect a continuous output?

[test_m2k-awg.zip](https://github.com/analogdevicesinc/libiio/files/1204883/test_m2k-awg.zip)

Thanks
Doug

lclausen-adi, 2017-08-07:
Hi,

The write function expects a bytearray (or something that can be converted to a bytearray) that contains the data with the right number of bits. E.g. for a 16-bit converter there should be two bytes per sample.

Can you try the following: AWGAwaveform = numpy.array(AWGAwaveform, dtype="int16")

lclausen-adi, 2017-08-07:
Looks like explicit conversion to bytearray is necessary: AWGAwaveform = bytearray(numpy.array(AWGAwaveform, dtype="int16"))

The other thing you set the 'sync' property to 1. This will prevent the DAC from starting to send data until the 'sync' flag is set to 0 again.

damercer, 2017-08-07:
That seems to be the right magic to make things work.

Thanks much for the info.

Doug

rgetz, 2020-01-31:
Closing - if you still need more help - let us know.

closed 2020-01-31 by rgetz