analogdevicesinc/libiio #113: M2K hardware triggering time out error?

damercer, 2017-08-13:
Hi:
I'm not sure if this issue should be in libiio or in the M2K firmware section but....
When using the hardware triggering functions on the M2K with Python (iio.py) everything is fine as long as the trigger condition happens before the context timeout limit is hit. But if it does not ( which could be a common occurrence ) an error is thrown. In my program (see code snip below ) I try to catch the error and turn off the triggering and request another refill of the buffer. But apparently the first error has done something that breaks everything and the second try ( with triggering off hopefully ) also fails.

It would seem to me that this triggering time out issue should he handled lower down in the software and done more gracefully and transparent to the higher level GUI. If the trigger event does not happen in some set time-out then gracefully send a filled buffer with what was acquired and set a triggered / not triggered flag that could be polled by the GUI.

Thanks

Doug

Code snip:

    #
    if OldShowSamples != SHOWsamples: # number of samples has changed
        del(TimeBuffer) # delete old buffer and make a new one
        TimeBuffer = iio.Buffer(m2k_adc, SHOWsamples, False)
    try: # catch error if buffer refill times out
        TimeBuffer.refill() # fill the buffer
        if TgInput.get() > 0: # check to see if triggering is on or not
            Is_Triggered = 1
        else:
            Is_Triggered = 0
    except:
        adc0.enabled = True
        adc1.enabled = True
        m2k_adc4_trigger.attrs["mode"].value = 'always' # turn off triggering
        m2k_adc5_trigger.attrs["mode"].value = 'always'
        m2k_adc6_trigger.attrs["logic_mode"].value = 'a_OR_b' # Should not matter
        print m2k_adc4_trigger.attrs["mode"].value
        print m2k_adc5_trigger.attrs["mode"].value
        # del(TimeBuffer) # delete old buffer and make a new one?
        print m2k_adc
        # TimeBuffer = iio.Buffer(m2k_adc, SHOWsamples, False)
        TimeBuffer.refill() # fill the buffer, this is line 2104 where it times out a second time
        Is_Triggered = 0
    x = TimeBuffer.read() # read the buffer

Shell snip:

Python 2.7.12 (v2.7.12:d33e0cf91556, Jun 27 2016, 15:19:22) [MSC v.1500 32 bit (Intel)] on win32
Type "copyright", "credits" or "license()" for more information.
>>> 
 RESTART: C:\Users\Doug\Documents\RPI\ADiscovery\ALM\libiio-test\alice-desktop-2.0.pyw 
No Init File Read
Library version: 0.10 (git tag: 4bb21be)

....

always
always
<iio.Device object at 0x02FF86F0>

Traceback (most recent call last):
  File "C:\Users\Doug\Documents\RPI\ADiscovery\ALM\libiio-test\alice-desktop-2.0.pyw", line 12743, 

in <module>
    Analog_In()
  File "C:\Users\Doug\Documents\RPI\ADiscovery\ALM\libiio-test\alice-desktop-2.0.pyw", line 1714, 

in Analog_In
    Analog_Time_In()
  File "C:\Users\Doug\Documents\RPI\ADiscovery\ALM\libiio-test\alice-desktop-2.0.pyw", line 2104, 

in Analog_Time_In
    TimeBuffer.refill() # fill the buffer
  File "C:\Users\Doug\Documents\RPI\ADiscovery\ALM\libiio-test\iio.py", line 533, in refill
    _buffer_refill(self._buffer)
  File "C:\Users\Doug\Documents\RPI\ADiscovery\ALM\libiio-test\iio.py", line 39, in _checkNegative
    raise OSError(-result, _strerror(-result))
OSError: [Errno 9] Bad file descriptor

lclausen-adi, 2017-08-17:
Hi,

Once a refill operation fails a buffer is in an error state from which it can't recover. Any future refill operation will also fail.

In your script you have commented out lines to create a new buffer. Do you get the same error when you enable them?

damercer, 2017-08-17:
I have not tired it lately but at first I did and it also caused an 
error. Don't remember if it was the same one. I can try again and get 
back to you.

Doug

damercer, 2017-08-17:
Adding back the lines to delete old buffer and make new buffer also caused an error, but a different one this time. Seems it no longer recognizes the device. Is there something else that needs to be reset/restarted here?

code snip:

    try: # catch error if buffer refill times out
        TimeBuffer.refill() # fill the buffer
        if TgInput.get() > 0: # check to see if triggering is not none
            Is_Triggered = 1
        else:
            Is_Triggered = 0
    except:
        adc0.enabled = True
        adc1.enabled = True
        m2k_adc4_trigger.attrs["mode"].value = 'always' # turn off triggering
        m2k_adc5_trigger.attrs["mode"].value = 'always'
        m2k_adc6_trigger.attrs["logic_mode"].value = 'a_OR_b'
        print m2k_adc4_trigger.attrs["mode"].value
        print m2k_adc5_trigger.attrs["mode"].value
        del(TimeBuffer) # delete old buffer and make a new one
        print m2k_adc
        TimeBuffer = iio.Buffer(m2k_adc, SHOWsamples, False)
        TimeBuffer.refill() # fill the buffer
        Is_Triggered = 0
    x = TimeBuffer.read() # read the buffer

Shell snip:

Python 2.7.12 (v2.7.12:d33e0cf91556, Jun 27 2016, 15:19:22) [MSC v.1500 32 bit (Intel)] on win32
Type "copyright", "credits" or "license()" for more information.
>>> 
 RESTART: C:\Users\Doug\Documents\RPI\ADiscovery\ALM\libiio-test\alice-desktop-2.0.pyw 
No Init File Read
Library version: 0.10 (git tag: 4bb21be)

.....

always
always
<iio.Device object at 0x032601D0>

Traceback (most recent call last):
  File "C:\Users\Doug\Documents\RPI\ADiscovery\ALM\libiio-test\alice-desktop-2.0.pyw", line 12898, in <module>
    Analog_In()
  File "C:\Users\Doug\Documents\RPI\ADiscovery\ALM\libiio-test\alice-desktop-2.0.pyw", line 1735, in Analog_In
    Analog_Time_In()
  File "C:\Users\Doug\Documents\RPI\ADiscovery\ALM\libiio-test\alice-desktop-2.0.pyw", line 2143, in Analog_Time_In
    TimeBuffer = iio.Buffer(m2k_adc, SHOWsamples, False)
  File "C:\Users\Doug\Documents\RPI\ADiscovery\ALM\libiio-test\iio.py", line 516, in __init__
    self._buffer = _create_buffer(device._device, samples_count, cyclic)
  File "C:\Users\Doug\Documents\RPI\ADiscovery\ALM\libiio-test\iio.py", line 33, in _checkNull
    raise OSError(err, _strerror(err))
OSError: [Errno 6] No such device or address
>>> 

Doug

lclausen-adi, 2017-08-23:
Took a while to track this down. The problem is with the implicit free of buffer. Even though you call del() something (probably the exception strackframe) still holds a reference to the old buffer and hence creating a new one fails. As a workaround:

```python
def capture():
    buf = iio.Buffer(m2k_adc, 1000, False)
    try:
        buf.refill()
        print("Got data")
        return True
    except:
        print("Timeout")
        return False

trigger_ch_a.attrs['mode'].value = 'analog'
if not capture(): # Timeout
    trigger_ch_a.attrs['mode'].value = 'always'
    capture() # Got data
```

pcercuei, 2017-08-24:
Here 'buf' will be freed by the garbage collector, I think, so you can't be sure it's been freed in the second capture() call.

damercer, 2017-08-24:
So is there a way to insure the garbage collector frees it up in time for the next time capture is called which is immediately. 

I've gone back to using my software triggering routine in the mean time.

DOug

pcercuei, 2018-02-15:
Would it be related to the bug described in https://github.com/analogdevicesinc/libiio/pull/141?

damercer, 2018-02-15:
I don't know. I've since loaded a later Libiio, not sure which version ( and latest M2K firmware ) but have not tried the hardware triggering in awhile. I'll test it again when I get some time.

Thanks

Doug

damercer, 2018-02-15:
I'm now using Library version: 0.14 (git tag: 17b73d3) with v0.18 M2K firmware and the program still returns an unrecoverable error when triggering is on and the trigger level is outside the input signal range in buffer.refill() function.

Doug

rgetz, 2020-01-13:
@adisuciu did we run into this problem with libm2k?

adisuciu, 2020-01-14:
Just tried this script with libm2k in python. Everything works just fine.

```
import libm2k
import numpy as np

ctx=libm2k.m2kOpen()
ain=ctx.getAnalogIn()
aout=ctx.getAnalogOut()
ain.setRange(0,-10,10)
ain.setSampleRate(1000000)
trig=ain.getTrigger()
trig.setAnalogSource(0)
trig.setAnalogLevel(0,10)
trig.setAnalogCondition(0,libm2k.RISING_EDGE_ANALOG)
trig.setAnalogDelay(0)
trig.setAnalogMode(0, libm2k.ANALOG)
ctx.setTimeout(100)
ain.enableChannel(0, True)

aout.setSampleRate(0, 750000)
aout.setSampleRate(1, 750000)
aout.enableChannel(0, True)
aout.enableChannel(1, True)

x=np.linspace(-np.pi,np.pi,1024)
buffer1=np.linspace(-2.0,2.00,1024)
buffer2=np.sin(x)

buffer = [buffer1, buffer2]

aout.setCyclic(True)
aout.push(buffer)

try:
    data=ain.getSamples(100)
    print("triggered")
except:
    print("no Trigger")
    trig.setAnalogMode(0,libm2k.ALWAYS)
    data=ain.getSamples(100)

print(data)
```

damercer, 2020-01-14:
I have not gone back and tried to do this in my libiio software but I was able to get the hardware triggering to work using my new version that uses libm2k functions.

So this is probably a non issue at this point.

Doug

closed 2020-01-14 by damercer