analogdevicesinc/libiio #114: iio_device_debug_attr_read_longlong not returning error code via network context

mfornero, 2017-08-16:
When using the `iio_device_debug_attr_read_longlong` API for an attribute that returns an error code, no error is produced via the network context, despite the debugfs layer reporting an error.

In this case, `iio_device_debug_attr_read_longlong` is called as part of `iio_device_reg_read`, for an address that always returns an error. When running via the network context, the return code from `iio_device_reg_read` is zero, despite the kernel log showing

```
[ 5021.046684]  mwipcore0:mmwr0: iio_debugfs_read_reg: read failed
```

Similarly, from the register via the shell using via debugfs, the above error is shown.

When breaking up the `iio_device_reg_read` function into its respective `iio_device_debug_attr_write_longlong` and `iio_device_debug_attr_read_longlong` calls, we can observe that it is `iio_device_debug_attr_read_longlong` that is not returning an error code correctly.

rgetz, 2017-08-17:
Matt:

Does it work on local context?

mfornero, 2017-08-17:
Haven't had a chance to test that-- I observed it in some code that primarily runs via the network context.

I'll see if I can spin up a test case for local shortly.

pcercuei, 2017-08-21:
When reading from the shell, the error is shown, but is the return code (variable $?) set to non-zero?

mfornero, 2017-08-21:
Great question-- looks like the kernel error is shown, but the return code is set to 0

```
zynq> cat direct_reg_access
[415206.121462]  mwipcore0:mmwr0: iio_debugfs_read_reg: read failed
0x0
[415206.127475]  mwipcore0:mmwr0: iio_debugfs_read_reg: read failed
[415206.133906]  mwipcore0:mmwr0: iio_debugfs_read_reg: read failed
zynq> echo $?
0
```

However, on a write, we see the correct behavior:

```
zynq> echo 0x0 0x0 > direct_reg_access
[415432.778751]  mwipcore0:mmwr0: iio_debugfs_write_reg: write failed
sh: write error: Permission denied
zynq> echo $?
1
```

The debugfs_reg_access function is here:
https://github.com/mathworks/xilinx-linux/blob/mathworks_psp_iiosdr_R17.1.1/drivers/misc/mathworks/mw_mm_iio_channel.c#L95

In this case, it is returning -EACCES, due to the mode being MW_MM_IO_MODE_DISABLED

Maybe some part of the kernel IIO layer is not correctly propagating the error in the case of a read?

lclausen-adi, 2017-08-21:
Yeah.

https://github.com/mathworks/xilinx-linux/blob/mathworks_psp_iiosdr_R17.1.1/drivers/iio/industrialio-core.c#L309-L317

mfornero, 2017-08-22:
I believe this should fix it-- I'll try to build / test later today

https://github.com/mfornero/linux/commit/5ae382797fc0304085af01d18c202d6dcfef705c

closed 2017-08-23 by lclausen-adi