analogdevicesinc/libiio #115: C# binding not implementing iio_context_get_attr

Manawyrm, 2017-08-20:
Hi,

the c# binding doesn't seem to implement a way to access the Context attributes, normally accessed via iio_context_get_attr.

rgetz, 2017-08-20:
The C# bindings were added before context attributes existed, so it's possible it wasn't updated. 

Thanks for pointing it out.

Manawyrm, 2017-08-20:
Off-Topic: 
I need to use the libad9361/ad9361_set_bb_rate function in C#. 
There isn't a binding for that atm; Would you recommend I create such a binding, or should I rather reimplement the functionality in C# nativly?

pcercuei, 2017-08-21:
Should be very easy to add.

lclausen-adi, 2017-08-21:
@Manawyrm looks like you have the best C# skills around here and an environment to test things, do you think you can send a patch?

rgetz, 2020-01-13:
iio_context_get_attr is still missing from bindings/csharp/Context.cs

rgetz, 2020-01-13:
either Dan or Adrian - can you have a look at this?

dNechita, 2020-01-14:
Yes

rgetz, 2020-02-07:
I think this is just a specific version of #364, so we will close this, and make sure it's addressed as part of that one. @Manawyrm feel free to comment there on what you think you need.

Thanks
-Robin

closed 2020-02-07 by rgetz