analogdevicesinc/libiio #116: C# binding - Method for getting raw IntPtr of device

Manawyrm, 2017-08-20:
Hello,

in order to use a binding of the libad9361 to control such a transceiver, I need the IntPtr device handle, which I currently can't get via a library method. 
At the moment I have to use reflection to get the pointer, which isn't really appropriate for production software... 

Generelly would libad9361-bindings for C# be very helpful. 

Thanks,
Tobias Mädel

pcercuei, 2017-08-21:
I'm not very keen on exposing private stuff outside the C# bindings...
Do you need all the functions of libad9361? It should be easy enough to re-implement parts of it in C# on top of the libiio C# bindings.

rgetz, 2020-01-13:
I am going to close this here, and move things over to libad9361-iio for c# bindings.

https://github.com/analogdevicesinc/libad9361-iio/issues/52

closed 2020-01-13 by rgetz