analogdevicesinc/libiio #118: Problem with 32 bit Python from issue #48 seems to be back?

damercer, 2017-09-06:
Hi With a recent version of libiio installed on my 64 bit ADI laptop running 32 bit Python 2.7 iio_info.py returns the following error with M2K plugged in:

Python 2.7.11 (v2.7.11:6d1b6a68f775, Dec  5 2015, 20:32:19) [MSC v.1500 32 bit (Intel)] on win32
Type "copyright", "credits" or "license()" for more information.
>>> 
============== RESTART: C:\My_Documents\ALM\iio-lib\iio_info.py ==============

Traceback (most recent call last):
  File "C:\My_Documents\ALM\iio-lib\iio_info.py", line 16, in <module>
    import iio
  File "C:\My_Documents\ALM\iio-lib\iio.py", line 49, in <module>
    _lib = _cdll.LoadLibrary('libiio.dll' if 'Windows' in _system() else 'libiio.so.0')
  File "C:\Python27\lib\ctypes\__init__.py", line 443, in LoadLibrary
    return self._dlltype(name)
  File "C:\Python27\lib\ctypes\__init__.py", line 365, in __init__
    self._handle = _dlopen(self._name, mode)
WindowsError: [Error 126] The specified module could not be found
>>> 

Not sure if it is the same problem exactly but something is missing for 32 bits.

I kind of need this working soon because I'm scheduled to do a demo at a local college next week.

Thanks

Doug

mhennerich, 2017-09-07:
Hi Doug,

On your PC delete C:\Windows\SysWOW64\libusb-1.0.dll
Then install the latest libiio release from here: https://github.com/analogdevicesinc/libiio/releases/latest

The issue seems to be that something has installed a debug enabled version of libusb which requires msvcr120d.dll. Our libiio installer doesn't overwrite libusb if it already exists.

-Michael

damercer, 2017-09-07:
No that didn't fix the problem. I did exactly what you said and when I run iio_info.py I get:

Python 2.7.11 (v2.7.11:6d1b6a68f775, Dec  5 2015, 20:32:19) [MSC v.1500 32 bit (Intel)] on win32
Type "copyright", "credits" or "license()" for more information.
>>> 
============ RESTART: C:\My_Documents\ALM\libiio-test\iio_info.py ============

Traceback (most recent call last):
  File "C:\My_Documents\ALM\libiio-test\iio_info.py", line 16, in <module>
    import iio
  File "C:\My_Documents\ALM\libiio-test\iio.py", line 62, in <module>
    use_errno = True, use_last_error = True)
  File "C:\Python27\lib\ctypes\__init__.py", line 365, in __init__
    self._handle = _dlopen(self._name, mode)
WindowsError: [Error 126] The specified module could not be found
>>> 

In a command window if I try and run iio_info.exe I get:

Microsoft Windows [Version 6.1.7601]
Copyright (c) 2009 Microsoft Corporation.  All rights reserved.

C:\Users\DMERCER>iio_info
Library version: 0.10 (git tag: 11b871b)
Compiled with backends: xml ip usb serial
Unable to create IIO context: Function not implemented

C:\Users\DMERCER>

It can't find any context ( i.e. the M2K ).

Must be something else???

Thanks for trying anyway.

Doug

pcercuei, 2017-09-07:
It uses the local backend by default which is not available on Windows. Try:
```iio_info -a```

damercer, 2017-09-08:
OK, that worked but iio_info.py still doesn't.

lclausen-adi, 2017-09-08:
If you just run a python shell in your C:\My_Documents\ALM\libiio-test\ and then run `import iio`, do you get any error messages about missing DLLs or something similar?

damercer, 2017-09-08:
Yes, In that directory I get the same error message. However in another directory with minimal stuff ( just iio.py ) typing import iio produces no errors. But running iio_info.py produces this different error:

Python 2.7.11 (v2.7.11:6d1b6a68f775, Dec  5 2015, 20:32:19) [MSC v.1500 32 bit (Intel)] on win32
Type "copyright", "credits" or "license()" for more information.
>>> import iio
>>> 
============== RESTART: C:\My_Documents\ALM\iio-lib\iio_info.py ==============
Library version: 0.10 (git tag: 4bb21be)

Traceback (most recent call last):
  File "C:\My_Documents\ALM\iio-lib\iio_info.py", line 55, in <module>
    main()
  File "C:\My_Documents\ALM\iio-lib\iio_info.py", line 20, in main
    ctx = iio.Context()
  File "C:\My_Documents\ALM\iio-lib\iio.py", line 667, in __init__
    self._context = _new_default()
  File "C:\My_Documents\ALM\iio-lib\iio.py", line 27, in _checkNull
    raise Exception("Null pointer")
Exception: Null pointer
>>> 

looks like it's finding the .dll but now runs into a different problem.

damercer, 2017-09-08:
The reason import iio worked in the other directory was that it happened to contain a copy of msvcr120.dll. When I changed the name of this file it reverted back to the original error.

So this would point to the fact that in the normal case it is not finding the msvcr120.dll in the default system path.

Here is the result if I type path in a command console:

Microsoft Windows [Version 6.1.7601]
Copyright (c) 2009 Microsoft Corporation.  All rights reserved.

C:\Users\DMERCER>path
PATH=C:\Python27\;C:\Python27\Scripts;C:\Windows\system32;C:\Windows;C:\Windows\
System32\Wbem;C:\Windows\System32\WindowsPowerShell\v1.0\;C:\Program Files\WIDCO
MM\Bluetooth Software\;C:\Program Files\WIDCOMM\Bluetooth Software\syswow64;C:\P
rogram Files\Intel\WiFi\bin\;C:\Program Files\Common Files\Intel\WirelessCommon\
;C:\CADTOOLS\ADI_CAD\lgb\win32\;c:\CADTOOLS\ADI_CAD\tc\win32\;c:\CADTOOLS\ADI_CA
D\adice5\win32\;C:\Program Files (x86)\Analog Devices\libsmu

C:\Users\DMERCER>

Doug

damercer, 2017-09-08:
Solution was to install the x86 version of the VC++ 2013 redistribution.
With this installed it all works now.

closed 2017-09-08 by damercer

mhennerich, 2017-09-08:
@pcercuei  -
Hi Paul,

shouldn't we install the **Visual C++ Redistributable Packages for Visual Studio 2013** 
along with the libiio installer?

-Michael

pcercuei, 2017-09-08:
Yes, IIRC the appveyor script copies that DLL to the ZIP files, however the Inno Setup script does not include it inside the installer. That would be nice to have.