analogdevicesinc/libiio #120: Arithmetic operations on a void pointer

ttrostel, 2017-09-11:
In the ad9361-iiostream.c example on lines 266 and 277 pdat is defined as a void* and therefore should not allow arithmetic operations.  I think you're supposed to cast it to something first.  Because the C Compiler in Xcode is a little bit less forgiving this code will not work on the MAC as it stands.  If I have the solution before you get to this I'll make sure to post it.

Thank you

ttrostel, 2017-09-11:
Line 252 can be changed to:
char *p_dat, *p_end;

Then line 266 can be changed to:
for (p_dat = (char *)iio_buffer_first(rxbuf, rx0_i); p_dat < p_end; p_dat += p_inc) {

and line 277 can be changed the same way.

mhennerich, 2017-09-13:
Fixed with this commit 7958803a9c08df94438ed6dd8c98309459b2de70
Thanks for pointing out.

closed 2017-09-13 by mhennerich