analogdevicesinc/libiio #125: JavaScript bindings

drom, 2017-10-23:
I want to use `iio` from my nodejs application. Do you have plans for the official JavaScript / NodeJS binding in the near future? Or I should go ahead and implement my own library?

lclausen-adi, 2017-10-23:
Hi,

We don't have any plans for adding this at the moment. But patches are always welcome :)

drom, 2018-06-26:
I am working on `libiio` bindings here: https://github.com/drom/node-iio

pcercuei, 2018-06-26:
Nice :)

pcercuei, 2018-11-06:
If you want, we can host the code inside the main repository, maybe that will give it more visibility.
I will close this bug report now that we effectively have JavaScript bindings.

closed 2018-11-06 by pcercuei

drom, 2018-11-13:
yes, I think it makes sense to move it into the main repository and mention its existence over the official LIBIIO channels. I am not using it at the moment, but other people may find it useful.

rgetz, 2020-04-15:
@drom : Did you want to make a pull request? Right now (we will just point to your repo - not yet live doc at https://rgetz.github.io/libiio/v0.19/index.html )

drom, 2020-04-16:
@rgetz I am actually OK with this Node.js IIO binding being a separate repo. I can move it into some `IIO` or `analogdevicesinc` org if there are concerns about maintainability. In the Node.js world, people appreciate small, "a single concern" modules. I also publish in on NPM as `libiio` where most of JS people will find it.

rgetz, 2020-04-17:
@drom - sounds great. I will also point to NPM as well... Thanks for the tip.