analogdevicesinc/libiio #126: we don't include LIBIIO_VERSION_PATCH

rgetz, 2017-10-24:
Today, the CMake is:

libiio-${VERSION}.g${LIBIIO_VERSION_GIT}

there is no way to make a release called 0.12.0, or 0.12.0-gxxxxxxx, it will always be 0.12.gxxxxxxx, where xxxxxxx is the git tag.

I'm just wondering if we should add this or just leave it.

-Robin

tfcollins, 2017-10-24:
I think if we had this option we would push out releases sooner at least.

pcercuei, 2017-10-24:
We can't change it now that we have API functions to get the version. It's too late.

rgetz, 2017-10-24:
Ok - iio_library_version_get_version is only major & minor as unsigned units, and git tag[8] as a char. 

You are right - we would need to break the API to include patch version.

lclausen-adi, 2017-10-24:
I think it is ok as it is at the moment. We are not going to maintain multiple versions in parallel anyway, so increasing the version number, even for just fixing things should be OK.

rgetz, 2017-10-24:
OK - no problem.

closed 2017-10-24 by rgetz