analogdevicesinc/libiio #129: local_read_dev_attr() zeroes the last byte regardless of the bufffer size

proski, 2017-11-09:
I'm using a device attribute to pass raw binary data to the client. I'm trying to read that data using libiio iio_device_attr_read() function. The last byte of the attribute is always set to zero, even if I provide enough space to put the zero after the binary data.

This code in local.c seems wrong to me

```
        if (ret > 0)
                dst[ret - 1] = '\0';
```

If `ret` is 60 and `len` is 64, the `'\0'` should go to `dst[61]`. Why damage data if the client provides enough space to hold the complete attribute data and the terminating null character?

lclausen-adi, 2017-11-09:
Hi,

Text attributes are terminated by a newline and this removes the newline. This function is meant for reading text attributes and it is not something we can change at this point since it would break all applications that do not expect the newline.

The best is probably to introduce a separate API for reading binary attributes that does not modify the content in any way.

pcercuei, 2017-11-09:
I thought this already existed, but apparently we don't have a API function for that. We have iio_device_attr_write_raw but no iio_device_attr_read_raw.

closed 2018-01-24 by lclausen-adi