analogdevicesinc/libiio #142: gh-pages is out of date

rgetz, 2017-12-31:
@pcercuei or @larsclausen 

https://github.com/analogdevicesinc/libiio/tree/gh-pages

has only 0.10 on it, since it is not automated. I can do this with travis-ci pretty easy.
https://gist.github.com/vidavidorra/548ffbcdae99d752da02

Do we want to update the gh-pages branch on every commit to master, or on tags (releases) only? ( would think releases only).

-Robin

pcercuei, 2018-01-01:
It should be done on releases only, yes. Automatizing the process would be a good idea.

mhennerich, 2018-01-05:
@rgetz  are you going to take care?

rgetz, 2018-01-05:
I think @AlexandraTrifan was going to have a look before the next release.

mhennerich, 2018-01-05:
if @AlexandraTrifan  wants to take a look that's ok. Longer term @commodo should be involved here too.

AlexandraTrifan, 2018-01-05:
I can take a look at this one if that's ok.

rgetz, 2018-01-05:
Thats fine - just assign the issue to yourself so everyone else knows you will be working on it.

PR #238 (2018-11-23): gh-pages: Commit autogenerated documentation on a tag
Closes #142
Only do this from one travis-ci instance
Only do this on tags
Commits happen from user "Autogenerated by Travis CI", with my email so
  if anyone wants to complain - they know who to contact.

Loosely based on https://gist.github.com/vidavidorra/548ffbcdae99d752da02
commit ddc3e4b4 (2018-11-23): gh-pages: Commit autogenerated documentation on a tag

Closes #142
Only do this from one travis-ci instance
Only do this on tags
Commits happen from user "Autogenerated by Travis CI", with my email so
  if anyone wants to complain - they know who to contact.

Loosely based on https://gist.github.com/vidavidorra/548ffbcdae99d752da02
commit 060d3fc8 (2019-05-23): gh-pages: Commit autogenerated documentation on a tag

Closes #142
Only do this from one travis-ci instance
Only do this on tags
Commits happen from user "Autogenerated by Travis CI", with my email so
  if anyone wants to complain - they know who to contact.

Loosely based on https://gist.github.com/vidavidorra/548ffbcdae99d752da02