analogdevicesinc/libiio #148: Add more architectures to downloads

markusr, 2018-02-14:
It would be great to have some more architectures as Downloads (.deb and .rpm). I think the most important one are for ARM platforms because of the many SBCs (raspberrypi, orangepi) 

- linux armhf
- linux arm64
- linux armel

Also a hint that the current packages for linux are all amd64 would be good.

rgetz, 2018-02-14:
@markusr - that is a great idea. 

The current packages are built on travis-ci or appveour. Do you have time to make a patch to do this? (Get travis to Install cross compile tools, and build?)

Thanks

markusr, 2018-02-14:
@rgetz  I will have a look what I can do. Not sure if I will succeed.

Which cross compile tools should be used? Linaro? / Debian?

mhennerich, 2018-02-15:
If you use a Debian based distro for ARM you can already get Debian packages:

https://packages.debian.org/sid/libiio0

markusr, 2018-02-19:
@mhennerich 
Yes, I know. The idea was more of getting the latest release.

I made some test but I encountered that cross compilation does not work, see https://github.com/analogdevicesinc/libiio/issues/27 .

We will first need to solve this issue.

rgetz, 2020-01-13:
There are armhf now, and upstream debian builds alot more...

@markusr : unless you have more that we can add to the CI, I will close this, and you can open a new one.

closed 2020-01-13 by rgetz