analogdevicesinc/libiio #149: iio context attributes

rgetz, 2018-02-21:
Thought this was puzzling, so I thought I would write it down.

I think local/remote descriptions aren't correct.

on host:
```
analog@imhotep:~/github/libiio$ iio_attr -a -C
Using auto-detected IIO context at URI "usb:3.2.5"
IIO context with 8 attributes:
local,kernel: 4.6.0-08783-g1cc7526
usb,idVendor: 0456
usb,idProduct: b673
usb,release: 2.0
usb,vendor: Analog Devices Inc.
usb,product: PlutoSDR (ADALM-PLUTO)
usb,serial: 104473222a8700071a002b001f4dd56fb8
```

however, on that same host:
```
analog@imhotep:~/github/libiio$ uname -a
Linux imhotep 4.9.0-5-amd64 #1 SMP Debian 4.9.65-3+deb9u2 (2018-01-04) x86_64 GNU/Linux
```

which doesn't match the above context local kernel. That version is the remote kernel (the Pluto device).
```
analog@imhotep:~/github/libiio$ sshanalog 
Warning: Permanently added '192.168.2.1' (ECDSA) to the list of known hosts.
Welcome to:
______ _       _        _________________
| ___ \ |     | |      /  ___|  _  \ ___ \
| |_/ / |_   _| |_ ___ \ `--.| | | | |_/ /
|  __/| | | | | __/ _ \ `--. \ | | |    /
| |   | | |_| | || (_) /\__/ / |/ /| |\ \
\_|   |_|\__,_|\__\___/\____/|___/ \_| \_|

v0.21
http://wiki.analog.com/university/tools/pluto
# uname -a
Linux pluto 4.6.0-08783-g1cc7526 #142 SMP PREEMPT Fri May 26 12:55:16 CEST 2017 armv7l GNU/Linux
```
a context attribute gets collected as "local", and then passed to the remote, but _local_ attributes are not  renamed to remote. (guessing)

pcercuei, 2018-02-22:
Well, that's expected. The prefixes for the context attributes correspond to the backends to which they apply. So "usb,..." attributes are for information related to the USB transport backend, and the "local,..." attributes are transfered from the device that creates the local backend. Then if for instance you want to check the version of the kernel that is used on the

pcercuei, 2018-02-22:
... device, you just need to read "local,kernel", independently of the platform the software is running on (PC or board).

rgetz, 2018-02-22:
So, "local" always refers to the location of the IIO device, not the location of the application actually connecting to something?

lclausen-adi, 2018-02-22:
Yes. Same way the devices that are part of the context are devices of the remote machine, not of the machine that the application is running on.

rgetz, 2018-02-22:
Ok - I will see if I can add anything to the doc later that captures this. 

Thanks

closed 2018-02-22 by rgetz